eclipse-zenoh/zenoh-pico #9: zenoh client

jiangfenghua123, 2020-12-11:
it does not support peer-to-peer communication.zenoh-pico targets constrained devices and offers a C API for pure clients.

how to test ?

I have  a zenoh  router, and a zenoh client ,client  cannot connect router!

Mallets, 2020-12-11:
Hi jiangfenghua123,

Can you please provide more details on your setup, the version of zenoh-pico and zenoh you are using, and how you are running them? How did you installed zenoh-pico and zenoh? 

Please post the output of zenohd running with the RUST_LOG=debug option enabled via command line so we can understand better what's going on.

jiangfenghua123, 2020-12-11:
https://github.com/eclipse-zenoh/zenoh 
install zenoh on linux arm64 board

https://github.com/eclipse-zenoh/zenoh-pico
install zenoh-pico  on linux amd64 pc

zenoh start  ./zenohd 
zenoh-pio  /zenoh-pico/build/examples$ ./zn_info
Openning session...
Please make sure one is running on your network!
Unable to open session!

Mallets, 2020-12-11:
Can you please laucnh ./zenohd as follows and post the output?

`RUST_LOG=debug ./zenohd`

Are the client and router running on the same machine?

jiangfenghua123, 2020-12-11:
user@localhost:~/zenoh/target/release$ RUST_LOG=debug ./zenohd
[2020-12-11T10:01:36Z DEBUG zenohd] zenohd v0.5.0-beta.5-89-gd1922e0
[2020-12-11T10:01:36Z DEBUG zenoh_util::lib_loader] Cannot search for libraries in /home/user/.zenoh/lib: No such file or directory (os error 2)
[2020-12-11T10:01:36Z DEBUG zenoh_router::plugins] Plugins to load: []
[2020-12-11T10:01:36Z DEBUG zenoh_util::lib_loader] Search for libraries libzplugin_*.so to load in ["/usr/local/lib", "/usr/lib", "/home/user/zenoh/target/release", "/home/user/zenoh/target/release"]
[2020-12-11T10:01:36Z DEBUG zenoh_util::lib_loader] Do not load plugin storages from "/home/user/zenoh/target/release/libzplugin_storages.so" : already loaded.
[2020-12-11T10:01:36Z DEBUG zenoh_util::lib_loader] Do not load plugin http from "/home/user/zenoh/target/release/libzplugin_http.so" : already loaded.
[2020-12-11T10:01:36Z DEBUG zenoh_router::plugins] Plugin storages loaded from /home/user/zenoh/target/release/libzplugin_storages.so
[2020-12-11T10:01:36Z DEBUG zenoh_router::plugins] Plugin http loaded from /home/user/zenoh/target/release/libzplugin_http.so
[2020-12-11T10:01:36Z DEBUG zenohd] Config: {"mode": "router", "peer": "", "add_timestamp": "true", "listener": "tcp/0.0.0.0:7447"}
[2020-12-11T10:01:36Z INFO  zenoh_router::runtime] Using PID: 324E135FEF5548D89784093595584348
[2020-12-11T10:01:36Z DEBUG zenoh_router::runtime::orchestrator] Listener tcp/0.0.0.0:7447 added
[2020-12-11T10:01:36Z INFO  zenoh_router::runtime::orchestrator] zenohd can be reached on tcp/192.168.10.75:7447
[2020-12-11T10:01:36Z INFO  zenoh_router::runtime::orchestrator] zenohd can be reached on tcp/192.168.1.238:7447
[2020-12-11T10:01:36Z INFO  zenoh_router::runtime::orchestrator] zenohd can be reached on tcp/172.17.0.1:7447
[2020-12-11T10:01:36Z DEBUG zenoh_router::runtime::orchestrator] UDP port bound to 224.0.0.224:7447
[2020-12-11T10:01:36Z DEBUG zenoh_router::runtime::orchestrator] Joined multicast group 224.0.0.224
[2020-12-11T10:01:36Z INFO  zenoh_router::runtime::orchestrator] zenohd listening scout messages on 224.0.0.224:7447
[2020-12-11T10:01:36Z DEBUG zenoh_router::runtime::orchestrator] UDP port bound to 192.168.10.75:55757
[2020-12-11T10:01:36Z DEBUG zenoh_router::plugins] Start plugin storages
[2020-12-11T10:01:36Z DEBUG zenoh_router::runtime::orchestrator] Waiting for UDP datagram...
[2020-12-11T10:01:36Z DEBUG zenoh_router::plugins] Start plugin http
[2020-12-11T10:01:36Z DEBUG zenoh_util::lib_loader] Cannot search for libraries in /home/user/.zenoh/lib: No such file or directory (os error 2)
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::broker] New face 0
[2020-12-11T10:01:36Z DEBUG zenoh] New workspace with prefix: Some(Path { p: "/@/router/324E135FEF5548D89784093595584348/plugin/storages/backend" })
[2020-12-11T10:01:36Z DEBUG zplugin_storages] Memory backend enabled
[2020-12-11T10:01:36Z DEBUG zenoh] New workspace with prefix: Some(Path { p: "/@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory" })
[2020-12-11T10:01:36Z DEBUG zenoh::workspace] eval on /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::resource] Register resource /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory for face 0
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::broker] New face 1
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/324E135FEF5548D89784093595584348 for face 1
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::broker] New face 2
[2020-12-11T10:01:36Z DEBUG zenoh::workspace] subscribe on storage/*
[2020-12-11T10:01:36Z INFO  tide::listener::tcp_listener] Server listening on http://0.0.0.0:8000
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::resource] Register resource /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory/storage/
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::resource] Register resource /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory/storage/*
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/memory/storage/* for face 0
[2020-12-11T10:01:36Z DEBUG zenoh::workspace] subscribe on /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/*
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::resource] Register resource /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::resource] Register resource /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/*
[2020-12-11T10:01:36Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/324E135FEF5548D89784093595584348/plugin/storages/backend/* for face 0

The client and router are not running on the same machine.

Mallets, 2020-12-11:
Ok, zenoh clients perform multicast scouting to discover where the router are and then to connect to them. If you are not running them on the same machine, are they at least running in the same LAN? This seems a problem on network configuration at your side. Can you please tell me what kind of LAN scenario do you have? How the different IP subnets are configured?

jiangfenghua123, 2020-12-11:
zenoh 192.168.10.75
zenoh client 192.168.10.194

zenoh-pico/build/examples$  ./zn_scout 
Scouting...
Did not find any zenoh process.

Mallets, 2020-12-11:
Zenoh uses multicast UDP messages to perform scouting, can you please check that multicast is properly working in your network? Perhaps using some tool like tcpdump or wireshark?

In the meanwhile, can you check that by specifying the locators on both zenoh router and zenoh client things work as expected?

`zenoh/target/release $ ./zenohd -l tcp/192.168.10.75:7447`
`zenoh-pico/build/examples $ ./zn_info tcp/192.168.10.75:7447`

jiangfenghua123, 2020-12-14:
./zn_info  tcp/192.168.10.75:7447
Openning session...
info_pid : 2E9082D8F9B06145
info_router_pid : 4236E62C84BB44EF9F68D39FDD4B836C

how to test zn_sub zn_pub?

jiangfenghua123, 2020-12-14:
some multicast UDP messages:
13:36:48.060508 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:36:56.060338 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:37:04.060290 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:37:12.060136 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:37:20.060094 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:37:28.059893 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2
13:37:36.060224 IP 192.168.10.75.50240 > 224.0.0.224.7447: UDP, length 2

Mallets, 2020-12-14:
Do you see any multicast UDP on 192.168.10.194? Do those machines have multiple network interfaces? If yes, can you please post the FULL network scenario? It might be that multicast discovery is not operating on the right network interface or multicast is not allowed in your network. Are the zenoh-router and zenoh-client running in a virtualized environment (e.g., Docker, VM, etc.) or directly on the OS?

In the meanwhile, you can test the zn_pub and zn_sub in the following way:

`zenoh/target/release $ ./zenohd -l tcp/192.168.10.75:7447`
The `-l tcp/192.168.10.75:7447` tells the router to listen on TCP port 7447 bound to IP address 192.168.10.75.

`zenoh-pico/build/examples $ ./zn_sub tcp/192.168.10.75:7447`
`zenoh-pico/build/examples $ ./zn_pub tcp/192.168.10.75:7447`
The  zn_sub and zn_pub are told to connect to the TCP port 7447 bound to IP address 192.168.10.75.
Same CLI argument can be passed to any other zenoh-pico example.

jiangfenghua123, 2020-12-14:
The zenoh-router and zenoh-client are running directly on the OS.

18:45:26.648178 IP jiang-XPS-15-9570.59882 > 239.255.255.250.1900: UDP, length 172
18:45:27.651184 IP jiang-XPS-15-9570.59882 > 239.255.255.250.1900: UDP, length 172
18:45:28.652327 IP jiang-XPS-15-9570.59882 > 239.255.255.250.1900: UDP, length 172
18:45:29.653240 IP jiang-XPS-15-9570.59882 > 239.255.255.250.1900: UDP, length 172
18:47:26.651213 IP jiang-XPS-15-9570.34438 > 239.255.255.250.1900: UDP, length 172
18:47:27.651676 IP jiang-XPS-15-9570.34438 > 239.255.255.250.1900: UDP, length 172
18:47:28.652163 IP jiang-XPS-15-9570.34438 > 239.255.255.250.1900: UDP, length 172
18:47:29.652909 IP jiang-XPS-15-9570.34438 > 239.255.255.250.1900: UDP, length 172

 tcp/192.168.2.1:7447 => tcp/192.168.2.10:57740

It might be that multicast discovery is not operating on right network port.

~/work/zenoh-pico/build/examples$ ./zn_sub tcp/192.168.2.1:7447
Openning session...
Please make sure one is running on your network!
Unable to open session!

Mallets, 2020-12-14:
Both zn_sub and zn_pub need to connect to a router. Zenoh-pico is pure-client implementation, so there is no peer-to-peer communication. All the traffic always needs to go through the router.
So make sure that you are running zenoh-router somewhere and pass the correct router locator to zn_sub and zn_pub as I posted in the previous reply where it is assumed that a router is running on tcp/192.168.10.75:7447.

jiangfenghua123, 2020-12-14:
1. mechine  arm64  fm1-Mac 192.168.2.1 
2. pc  amd64 port1 192.168.2.10
They are all connected to the switch.

:~/work/zenoh-pico/build/examples$ ./zn_sub tcp/192.168.2.1:7447
Openning session...
Please make sure one is running on your network!
Unable to open session!

~/work/zenoh-pico/build/examples$ ./zn_pub tcp/192.168.2.1:7447
Openning session...
Please make sure one is running on your network!
Unable to open session!

Mallets, 2020-12-14:
Is the zenoh-router running at tcp/192.168.2.1:7447?

jiangfenghua123, 2020-12-14:
yes.
~/zenoh/target/release$  RUST_LOG=debug ./zenohd -l tcp/192.168.2.1:7447
[2020-12-14T10:41:35Z DEBUG zenohd] zenohd v0.5.0-beta.5-89-gd1922e0
[2020-12-14T10:41:35Z DEBUG zenoh_util::lib_loader] Cannot search for libraries in /home/user/.zenoh/lib: No such file or directory (os error 2)
[2020-12-14T10:41:35Z DEBUG zenoh_router::plugins] Plugins to load: []
[2020-12-14T10:41:35Z DEBUG zenoh_util::lib_loader] Search for libraries libzplugin_*.so to load in ["/usr/local/lib", "/usr/lib", "/home/user/zenoh/target/release", "/home/user/zenoh/target/release"]
[2020-12-14T10:41:35Z DEBUG zenoh_util::lib_loader] Do not load plugin storages from "/home/user/zenoh/target/release/libzplugin_storages.so" : already loaded.
[2020-12-14T10:41:35Z DEBUG zenoh_util::lib_loader] Do not load plugin http from "/home/user/zenoh/target/release/libzplugin_http.so" : already loaded.
[2020-12-14T10:41:35Z DEBUG zenoh_router::plugins] Plugin storages loaded from /home/user/zenoh/target/release/libzplugin_storages.so
[2020-12-14T10:41:35Z DEBUG zenoh_router::plugins] Plugin http loaded from /home/user/zenoh/target/release/libzplugin_http.so
[2020-12-14T10:41:35Z DEBUG zenohd] Config: {"peer": "", "listener": "tcp/192.168.2.1:7447", "mode": "router", "add_timestamp": "true"}
[2020-12-14T10:41:35Z INFO  zenoh_router::runtime] Using PID: 4E3313562C3843BD9D938C375378A71E
[2020-12-14T10:41:35Z DEBUG zenoh_router::runtime::orchestrator] Listener tcp/192.168.2.1:7447 added
[2020-12-14T10:41:35Z INFO  zenoh_router::runtime::orchestrator] zenohd can be reached on tcp/192.168.2.1:7447
[2020-12-14T10:41:35Z DEBUG zenoh_router::runtime::orchestrator] UDP port bound to 224.0.0.224:7447
[2020-12-14T10:41:35Z DEBUG zenoh_router::runtime::orchestrator] Joined multicast group 224.0.0.224
[2020-12-14T10:41:35Z INFO  zenoh_router::runtime::orchestrator] zenohd listening scout messages on 224.0.0.224:7447
[2020-12-14T10:41:35Z DEBUG zenoh_router::runtime::orchestrator] UDP port bound to 192.168.2.1:45773
[2020-12-14T10:41:35Z DEBUG zenoh_router::plugins] Start plugin storages
[2020-12-14T10:41:35Z DEBUG zenoh_router::runtime::orchestrator] Waiting for UDP datagram...
[2020-12-14T10:41:35Z DEBUG zenoh_router::plugins] Start plugin http
[2020-12-14T10:41:35Z DEBUG zenoh_util::lib_loader] Cannot search for libraries in /home/user/.zenoh/lib: No such file or directory (os error 2)
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::broker] New face 0
[2020-12-14T10:41:35Z DEBUG zenoh] New workspace with prefix: Some(Path { p: "/@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend" })
[2020-12-14T10:41:35Z DEBUG zplugin_storages] Memory backend enabled
[2020-12-14T10:41:35Z DEBUG zenoh] New workspace with prefix: Some(Path { p: "/@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory" })
[2020-12-14T10:41:35Z DEBUG zenoh::workspace] eval on /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::resource] Register resource /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory for face 0
[2020-12-14T10:41:35Z DEBUG zenoh::workspace] subscribe on storage/*
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::resource] Register resource /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory/storage/
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::resource] Register resource /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory/storage/*
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/memory/storage/* for face 0
[2020-12-14T10:41:35Z DEBUG zenoh::workspace] subscribe on /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/*
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::resource] Register resource /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::resource] Register resource /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/*
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/4E3313562C3843BD9D938C375378A71E/plugin/storages/backend/* for face 0
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::broker] New face 1
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::broker] New face 2
[2020-12-14T10:41:35Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/4E3313562C3843BD9D938C375378A71E for face 1
[2020-12-14T10:41:35Z INFO  tide::listener::tcp_listener] Server listening on http://0.0.0.0:8000
[2020-12-14T10:43:38Z DEBUG zenoh_protocol::link::tcp] Accepted TCP connection on 192.168.2.1:7447: 192.168.2.10:57740
[2020-12-14T10:43:38Z DEBUG zenoh_protocol::session::manager] New session opened with B139F9C60ECBD2AE: whatami 4, lease 10s, keep alive 1000, sn resolution 268435456, initial sn tx 39680302, initial sn rx 149330566
[2020-12-14T10:43:38Z DEBUG zenoh_router::routing::broker] New face 3
[2020-12-14T10:43:38Z DEBUG zenoh_protocol::session::initial] New session link established from B139F9C60ECBD2AE: tcp/192.168.2.1:7447 => tcp/192.168.2.10:57740
[2020-12-14T10:43:38Z DEBUG zenoh_protocol::session::channel] Closing the session with peer: B139F9C60ECBD2AE
[2020-12-14T10:43:38Z DEBUG zenoh_router::routing::broker] Close face 3

Mallets, 2020-12-14:
Ok, the link is closed during the opening phase, which is weird. Can you please confirm that you are running both zenoh and zenoh-pico from the master branch from the github repo (and use the latest commits)?

jiangfenghua123, 2020-12-14:
yes , they are the master branch  from the github repo  (and use the latest commits).

Mallets, 2020-12-14:
Are you using one of the latest Mac with arm-based CPU?

jiangfenghua123, 2020-12-14:
four 64-bit Arm® Cortex®-A72 cores

Mallets, 2020-12-14:
But is the latest Mac or not? Which OS are you running?

jiangfenghua123, 2020-12-14:
linux  ubuntu 20.04

Mallets, 2020-12-14:
Ok, thanks. Ubuntu 20.04 32bits or 64bits?

jiangfenghua123, 2020-12-14:
zenoh client:
/work/zenoh-pico/build/examples$ ./zn_info  tcp/192.168.2.1:7447
Openning session...
info_pid : 43AA527F98DCC721
info_router_pid : A0F69356F48C413A9F40042FF0D432FE
zenoh router:
[2020-12-14T14:00:56Z DEBUG zenoh_protocol::link::tcp] Accepted TCP connection on 192.168.2.1:7447: 192.168.2.10:60416
[2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::manager] New session opened with 8CBCC75322C01D24: whatami 4, lease 10s, keep alive 1000, sn resolution 268435456, initial sn tx 208538509, initial sn rx 105649593
[2020-12-14T14:00:56Z DEBUG zenoh_router::routing::broker] New face 5
[2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::initial] New session link established from 8CBCC75322C01D24: tcp/192.168.2.1:7447 => tcp/192.168.2.10:60416
[2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::channel] Closing the session  with peer: 8CBCC75322C01D24
[2020-12-14T14:00:56Z DEBUG zenoh_router::routing::broker] Close face 5
###   _peer not client_

jiangfenghua123, 2020-12-14:
Ubuntu 20.04 arm64

Mallets, 2020-12-14:
> zenoh client:
> /work/zenoh-pico/build/examples$ ./zn_info tcp/192.168.2.1:7447
> Openning session...
> info_pid : 43AA527F98DCC721
> info_router_pid : A0F69356F48C413A9F40042FF0D432FE
> zenoh router:
> [2020-12-14T14:00:56Z DEBUG zenoh_protocol:🔗:tcp] Accepted TCP connection on 192.168.2.1:7447: 192.168.2.10:60416
> [2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::manager] New session opened with 8CBCC75322C01D24: whatami 4, lease 10s, keep alive 1000, sn resolution 268435456, initial sn tx 208538509, initial sn rx 105649593
> [2020-12-14T14:00:56Z DEBUG zenoh_router::routing::broker] New face 5
> [2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::initial] New session link established from 8CBCC75322C01D24: tcp/192.168.2.1:7447 => tcp/192.168.2.10:60416
> [2020-12-14T14:00:56Z DEBUG zenoh_protocol::session::channel] Closing the session with peer: 8CBCC75322C01D24
> [2020-12-14T14:00:56Z DEBUG zenoh_router::routing::broker] Close face 5
> 
> ### _peer not client_

The term peer in the log is used also to refer to client, i.e., simply the other endpoint. This should be improved in the logs indeed. In any case, this is not the source of the problem. We will try to replicate the scenario, in any case the more information you provide to us the better.

Mallets, 2020-12-15:
Can you please try with the following CLI commands? There were a couple of arguments missing in the previous posts.

`zenoh/target/release $ RUST_LOG=debug ./zenohd -l tcp/192.168.2.1:7447`
Tells the zenoh router to list on TCP port 7447 and IP address 192.168.2.1

`zenoh-pico/build/examples $ ./zn_sub /demo/example/**  tcp/192.168.2.1:7447`
Tells the zenoh-pico subscriber to connect to TCP port 7447 and IP address 192.168.2.1 and subscribe to '/demo/example/**'. You can replace the key selector according to [zenoh abstractions](http://zenoh.io/docs/manual/abstractions/).

`zenoh-pico/build/examples $ ./zn_pub /demo/example/zenoh-pico-pub "Pub from pico!" tcp/192.168.2.1:7447`
Tells the zenoh-pico publisher to connect to TCP port 7447 and IP address 192.168.2.1 and publish on '/demo/example/zenoh-pico-sub' the valu "Pub from pico!". You can replace both the key and the value as you please.

jiangfenghua123, 2020-12-15:
ok, client is working.

Mallets, 2020-12-15:
Great. I'm gonna close this issue.
Two related issues have been open related to the required improvements in zenoh-pico:
-  Improve network interface selection for multicast scouting #10 
-  Improve ergonomics of CLI arguments in examples #11  

In case of additional enquiries on how to run some examples please use zenoh [gitter channel](https://gitter.im/atolab/zenoh).

closed 2020-12-15 by Mallets