eclipse-zenoh/zenoh-pico #14: Add support for Zephyr

esteve, 2021-02-12:
Zephyr is an RTOS that has a mostly-POSIX layer. zenoh-pico is mostly abstracted and platform-agnostic, but there are still parts of the codebase that expect a Unix-like operating system (e.g. `getifaddrs` in `src/net/unix/network.c`).

esteve, 2021-02-12:
@Mallets can you assign this ticket to me? Thanks.

Mallets, 2021-02-13:
@esteve there is already the issue #8 which is very related to the porting to Zephyr. Are you fine if assign that issue to you as well since it is very related?

Mallets, 2021-05-18:
@esteve I believe this issue was closed by [PR24](https://github.com/eclipse-zenoh/zenoh-pico/pull/24). We should be able to close the issue now. Anything still pending?

esteve, 2021-05-18:
@Mallets nothing really important. There's the CI improvements I mentioned in another ticket, but that discussion in the PLATFORMIO forums hasn't moved forward.

Mallets, 2021-05-18:
Ok, since the support for Zephyr has already been merged I'll close this issue. I've opened a new specific issue #29 to keep track of the PlatformIO matter.

closed 2021-05-18 by Mallets