eclipse-zenoh/zenoh-pico #35: examples core dumps

vortex314, 2021-07-05:
Compilation and make went fine
./zn_sub core dumps when ./zn_pub starts emitting 
```
Segmentation fault (core dumped)
lieven@pcpav2:~/workspace/zenoh-pico/build/examples$ gdb zn_sub
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.                    
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from zn_sub...
--Type <RET> for more, q to quit, c to continue without paging--
(No debugging symbols found in zn_sub)
(gdb) run
Starting program: /home/lieven/workspace/zenoh-pico/build/examples/zn_sub 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".                                                                
Openning session...
[New Thread 0x7ffff7d68700 (LWP 937746)]
[New Thread 0x7fffef567700 (LWP 937747)]
Declaring Subscriber on '/demo/example/**'...

Thread 2 "zn_sub" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7d68700 (LWP 937746)]
0x00007ffff7fb41b2 in _z_iosli_read ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
(gdb) bt
#0  0x00007ffff7fb41b2 in _z_iosli_read ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#1  0x00007ffff7fba64c in _zn_zenoh_message_decode_na ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#2  0x00007ffff7fba866 in _zn_zenoh_message_decode ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#3  0x00007ffff7fbb703 in _zn_frame_decode_na ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#4  0x00007ffff7fbb7fe in _zn_frame_decode ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#5  0x00007ffff7fbbc8b in _zn_session_message_decode_na ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#6  0x00007ffff7fbbf7d in _znp_read_task ()
   from /home/lieven/workspace/zenoh-pico/build/libs/libzenohpico.so
#7  0x00007ffff7d75609 in start_thread (arg=<optimized out>)
    at pthread_create.c:477
#8  0x00007ffff7eb1293 in clone ()
--Type <RET> for more, q to quit, c to continue without paging--
   64/clone.S:95                                                     
(gdb)                        
```
While running zenoh in docker 
```
lieven@pcpav2:~/workspace/zenoh-pico$ docker run --init --net host eclipse/zenoh 
 * Starting: /zenohd 
[2021-07-05T21:15:46Z INFO  zenoh::net::runtime] Using PID: 67D34C027F50460795CFDAF56380AA86
[2021-07-05T21:15:46Z INFO  zenoh::net::runtime::orchestrator] zenohd can be reached on tcp/192.168.0.238:7447
[2021-07-05T21:15:46Z INFO  zenoh::net::runtime::orchestrator] zenohd can be reached on tcp/172.17.0.1:7447
[2021-07-05T21:15:46Z INFO  zenoh::net::runtime::orchestrator] zenohd listening scout messages on 224.0.0.224:7447
[2021-07-05T21:15:46Z INFO  tide::server] Server listening on http://0.0.0.0:8000
```
Any suggestions ?

Mallets, 2021-07-06:
Hi @vortex314 !

It seems that you are using zenoh-pico from Git (master) and zenoh from Docker (0.5.0.beta8). Those two versions are not compatible with each other. Either you pick a version from the [Eclipse download area](https://download.eclipse.org/zenoh/) for both zenoh and zenoh-pico or you compile both from Git (master).

vortex314, 2021-07-06:
Ok, will try that. Zenoh features look very much what I need to integrate micro-controllers into a bigger robotic system. Keep up the good work !

Mallets, 2021-07-06:
Cool! Please do not hesitate to raise additional questions also on [zenoh's Gitter channel](https://gitter.im/atolab/zenoh).

closed 2021-08-11 by Mallets