eclipse-zenoh/zenoh-pico #36: cmake: some options are not option(..)s

gavanderhoorn, 2021-07-06:
The current `CMakeLists.txt` has a few `option(..)`s to control things like whether Linux packages get built and whether the examples are compiled:

https://github.com/eclipse-zenoh/zenoh-pico/blob/913ee7caaac97a648b1fc1dd0ed75b12be855208/CMakeLists.txt#L41-L43

Some other options which would seem to make sense as `option(..)`s are not `option(..)`s though:

https://github.com/eclipse-zenoh/zenoh-pico/blob/913ee7caaac97a648b1fc1dd0ed75b12be855208/CMakeLists.txt#L32

https://github.com/eclipse-zenoh/zenoh-pico/blob/913ee7caaac97a648b1fc1dd0ed75b12be855208/CMakeLists.txt#L37

https://github.com/eclipse-zenoh/zenoh-pico/blob/913ee7caaac97a648b1fc1dd0ed75b12be855208/CMakeLists.txt#L45-L49

Is this on purpose?

I noticed that #19 and #21 have improved the `CMakeLists.txt`, but did not use `option(..)` for the items identified above. Perhaps that was for a reason, but I don't seem to find it in the reviews.

Mallets, 2021-07-07:
Hi @gavanderhoorn !

The `CMakeLists.txt` was ported by an old version of zenoh and we might have missed to adapt few things. Could you submit a PR with the proposed changes?

closed 2021-09-22 by Mallets

gavanderhoorn, 2021-09-22:
Thanks guys.

Apologies for not submitting a PR.