eclipse-zenoh/zenoh-pico #56: Segmentation fault on zn_sub zn_pub examples

frankplus, 2021-12-10:
Hi,
zn_pub is throwing a segmentation fault 

**How to reproduce**: 
run zenohd and zn_sub
run zn_pub 

the logs are as follows:
```console
$ ./zn_sub
Openning session...
Declaring Subscriber on '/demo/example/**'...
```

```console
$ ./zn_pub
Openning session...
Declaring Resource '/demo/example/zenoh-pico-pub' => RId 1
Declaring Publisher on 1
Writing Data ('1': '[   0] Pub from pico!')...
Writing Data ('1': '[   1] Pub from pico!')...
Segmentation fault (core dumped)
```

Running on Ubuntu 20.04
Zenohd version: zenoh router v0.5.0-beta.8-326-g70d7b22

Mallets, 2021-12-10:
Which version of zenoh-pico are you using? Did you compile it from master?

frankplus, 2021-12-10:
zenoh-pico was compiled from master

zenoh router was installed using apt install zenoh, but also compiling from zenoh master is giving the same error

Mallets, 2021-12-13:
I've just replicated the same scenario and I am not able to reproduce the segfault.

Can you please try the following?
```sh
git clone https://github.com/eclipse-zenoh/zenoh.git
cd zenoh
cargo update
cargo build
```

Then:
```sh
git clone of https://github.com/eclipse-zenoh/zenoh-pico.git
cd zenoh-pico
BUILD_TYPE=DEBUG make
```

In a first terminal:
```sh
cd zenoh
RUST_LOG=debug ./target/release/zenohd
```

Make sure you have `gdb` or `lldb` installed on your machine.

In a second terminal:
```sh
cd zenoh-pico
lldb ./build/examples/zn_sub
```
Start the subscriber from `gdb` or `lldb` command prompt:
```sh
(lldb) run
```

In a third terminal:
```sh
cd zenoh-pico
lldb ./build/examples/zn_pub
```
Start the publisher from `gdb` or `lldb` command prompt:
```sh
(lldb) run
```

When segfault happens, post the backtrace:
```sh
(lldb) bt
```

Finally, please also provide more information on your environment (e.g., Linux, MaxOSX, etc.), which distro you are using, its version, and what compiler you are using.

Thanks for your help.

frankplus, 2021-12-14:
I am in a new ubuntu workstation and the error is not appearing anymore.
unfortunately I don't have anymore the workstation where the error appeared, so I cannot try to debug

closed 2021-12-14 by frankplus