eclipse-zenoh/zenoh-pico #75: Segmentation Fault on zn_sub and zn_pub examples with external server

vara-chigurupati, 2022-03-31:
Hello,
When  I run the  `zn_sub` and `zn_pub` to connect to external server, they are failing with  **Unable to open session!.**

> **Steps to reproduce:**
> Zenoh server(`zenohd`) running on different machine, with DNS `zenoh.one.tech` on port `5353`.
> and  `zn_sub` and `zn_pub` with command line arguments as below.

> ```
> ./zn_pub /demo/example/zenoh-pico-pub hi udp/zenoh.one.tech:5353
> Openning session...
> Unable to open session!
> ```
> 
> ```
> ./zn_sub /demo/example/** udp/zenoh.one.tech:5353
> Openning session...
> Unable to open session!
> ```

Running on Ubuntu 20.04
zenoh-pico was compiled from master

cguimaraes, 2022-04-01:
Which version of zenohd are you using? Mismatched versions are one of possible the reasons for being unable to open the session.

vara-chigurupati, 2022-04-01:
Zenohd is 0.5.0-beta.9. 
Moved the zenoh-pico also to `0.5.0-beta.9` version. Then getting the segmentation fault.

cguimaraes, 2022-04-01:
I recommend you using the zenoh-pico from master branch as there were significant improvements since 0.5.0-beta.9. That means that you need either to compile the zenohd from master branch or to use the nightly docker image.

In any case, the release of zenohd and zenoh-pico in their version 0.5.0-beta.9 are compatible, thus you should not be getting the segmentation fault out of the examples. Can you share any backtrack? In any case, the issue might be already fixed in the master branch.

cguimaraes, 2022-04-05:
I am unable to replicate the issue with Zenoh-Pico (`0.5.0-beta.9`) and Zenohd (`0.5.0-beta.9`).

@vara-chigurupati would it be possible to share more information about your environment and the steps/commands you are executing?

vara-chigurupati, 2022-04-07:
Today I tried again on zenoh-pico `0.5.0-beta.9` and fortunately able to run successfully.

Not sure the reason, nothing different this time. 
Anyway Thank you so much for your response.

closed 2022-04-11 by cguimaraes