eclipse-zenoh/zenoh-pico #81: Potential double free in read task using multicast peer-mode

spiderkeys, 2022-04-14:
### Describe the bug

Observation:
When running a simple ping-pong example between two MCUs running zenoh-pico in multicast peer mode for some period of time (usually around 3 minutes, with a ping rate of 40 Hz), a bus fault is thrown and it appears a double free occurs in zenoh's read task:

[image]

I am very hesitant to make any judgement as to where the root cause of this fault comes from right now. It could very well be an issue on our side with our custom network driver, but I thought I would share this stack trace to see if it might trigger some thoughts on your side. I do see that there are a couple of commits past the commit we are working from that fix things like potential race conditions and changing write buffer index behaviors - not sure if it might be related.

Setup:
- Using Zephyr 3.0.99
- Network interface being used is a custom IP-over-serial driver in our stack
- Correct multicast and unicast behaviors tested separately from zenoh-pico
- Devices: 2x Nucleo STM32L496ZG-P talking to each other directly over ff03::01 multicast

 

### To reproduce

- Zenoh Commit: 3587bcf528bbb99f96051b7ea1dae5a59abf4d36
 1. Set up one device to act as publisher, publishing a simple counter:
 ```c
uint8_t counter = 0;
    while( true )
    {
        sprintf(buf, "%d", counter);
        zn_write_ext(s, zn_rname(uri), (const uint8_t *)buf, strlen(buf), Z_ENCODING_DEFAULT, Z_DATA_KIND_DEFAULT, zn_congestion_control_t_BLOCK);
        counter++;
        usleep( 25000 );
    }
```
2. Set up the other device to act as a subscriber and publisher, publishing a reply upon receiving the other device's counter message:
```c
void data_handler(const zn_sample_t *sample, const void *arg)
{
    (void)arg;

    /* The "consumer" returns an ACK back to the "producer". In this case both devices are Zenoh publishers and subscribers */
    sprintf(buf, "%s", ack);
    zn_write_ext(s, zn_rname(uri), (const uint8_t *)buf, strlen(buf), Z_ENCODING_DEFAULT, Z_DATA_KIND_DEFAULT, zn_congestion_control_t_BLOCK);

    LOG_INF(">> [Subscription listener] Received (%.*s, %.*s)\n",
           (int)sample->key.len, sample->key.val,
           (int)sample->value.len, sample->value.val);
}
```
3. Wait a few minutes until the usage fault occurs, producing the included stack trace.

### System info

Development platform: Ubuntu 20.04 x86_64
Zenoh Commit: 3587bcf528bbb99f96051b7ea1dae5a59abf4d36
Zephyr Version: 3.0.99 (with our custom network driver + OpenThread implementing the IPv6 interface)

MCU: Nucleo STM32L496ZG-P

spiderkeys, 2022-04-14:
One thought/question we have is whether or not it is currently ok from a thread/task safety perspective to call zn_write_ext() within the subscription callback executed within the read task.

cguimaraes, 2022-04-23:
> One thought/question we have is whether or not it is currently ok from a thread/task safety perspective to call zn_write_ext() within the subscription callback executed within the read task, as is currently done in the example producing the issue.

Doing this is not an issue.
In terms of concurrency, it is safe to do it either w.r.t. the zenoh session and TX/RX operations.

I am still trying to replicate the issue. I will come back with updates as soon as I get it.

cguimaraes, 2022-04-26:
@spiderkeys by any chance are you issuing any zenoh query during your tests?

From your backtrace, it seems a zenoh message containing a reply context is being freed.
However, such decorator is just being used for DATA messages whenever retrieving data from a query or for the UNIT message signaling the final reply.

spiderkeys, 2022-05-06:
Hi @cguimaraes,

Sorry for the delayed reply. We ended up tracking this down just recently to a bug in our custom network driver where packet buffer resources were not always being reliably returned to the network stack after upstream consumption. We fixed this and have never seen this error again, and are pretty confident that Zenoh just happened to be calling the network APIs that triggered the issue (we also saw it happen when using plain udp socket APIs in zephyr, which Zenoh used under the hood).

Going to close this issue. Thanks for taking the time to look into things on your side, and sorry for the red herring!

closed 2022-05-06 by spiderkeys