eclipse-zenoh/zenoh-pico #106: Assign IP address of interface to imr_interface member of ip_mreq struct

philhoeb, 2022-08-30:
### Describe the feature

Hi,

following test setup does not work:
- zn_peer_pub.c running on microcontroller connected to en1 interface of Linux workstation
- zn_peer_sub.c running on Linux workstation

The problem is that the subscriber listens on en0 interface for multicast messages.
This is due to the INADDR_ANY assignment to mreq.imr_interface.s_addr in zenoh-pico/src/system/unix/network.c.
"imr_address is the address of the local interface with which the system should join the multicast group; if it is equal to        INADDR_ANY, an appropriate interface is chosen by the system." - https://man7.org/linux/man-pages/man7/ip.7.html
"If this member specifies an IPv4 address of 0.0.0.0, the default IPv4 multicast interface is used." - https://docs.microsoft.com/en-us/windows/win32/api/ws2ipdef/ns-ws2ipdef-ip_mreq

To be able to handle multiple network interfaces, it would be better to pass the address of the respective interface to _zn_listen_udp_multicast() and _zn_close_udp_multicast().

Thanks!

cguimaraes, 2022-08-30:
@philhoeb thanks for noting this issue.

I have provided a quick fix as a draft PR (https://github.com/eclipse-zenoh/zenoh-pico/pull/107) to fix the behavior you describe. It still needs further testing, but it will take some days since I am currently out of office.
But in the meantime, if you can check if it fixes things on your side we would appreciate.

Also, it seems to me that you are using the `master` branch. I would advice you to use the `api-changes` branch until our next release, since it contains a big chunk of improvements to be release next month that were not yet merged into the `master` branch.

philhoeb, 2022-09-01:
@cguimaraes your changes fix this issue on my side.
I have tested peer to peer communication on interface en1 (non default multicast interface) when both subscriber and publisher run on Linux, as well as one of them running on a microcontroller connected to the interface. It works in both cases.

Thank you for this quick fix and also the hint regarding the api-changes branch!

commit b6e04b2e (2022-09-09): Fix #106 . Assign IP address of interface to imr_interface member of ip_mreq struct (#107)

* Avoid double free upon failed multicast link opening

* Properly set imr_interface value

* Add mode as argument for examples

* Move get ip from iface as an utility function
closed 2022-09-09 by cguimaraes