eclipse-zenoh/zenoh-pico #125: Incompatibility with the zenoh network router (zenohd)

philhoeb, 2022-09-28:
### Describe the bug

Hi,

the Unix pub/sub examples of the zenoh-pico master branch work perfectly well with the Docker image of the zenoh network router ($ docker run --init --net=host eclipse/zenoh:master). However, when running the Mbed pub example on a STM32 microcontroller program execution gets stuck when opening the zenoh session (z_open()). The STM32 microcontroller transmits the zenoh Init message but the network router answers with a zenoh Close message.

### To reproduce

1. Start the zenohd Docker image
2. Start the Mbed pub example
3. Inspect packets in wireshark

### System info

Platform: Ubuntu 20.04
Microcontroller: STM32 B-L475E-IOT01A

philhoeb, 2022-09-28:
It seems the problem is the Peer ID being zero. I have to look into the Mbed random number generator...
[image]
[image]

philhoeb, 2022-09-28:
Seeding randLIB solves the problem (e.g. randLIB_seed_random();)

cguimaraes, 2022-09-29:
Hi @philhoeb,
Thank you for identifying this issue on our examples.
In fact, the user of zenoh-pico libraries must initialize the seed for the random number generator. We decided in this direction so that the user, if desired, can obtain the same sequence of pseudo-random numbers between runs, thus allowing replicable analysis.

I have just pushed a PR to fix it in the examples: https://github.com/eclipse-zenoh/zenoh-pico/pull/127

commit 5482aacd (2022-09-29): Fix #125 . Init seed for random in mbed examples (#127)
closed 2022-09-29 by cguimaraes