eclipse-zenoh/zenoh-pico #129: FreeRTOS port

philhoeb, 2022-10-27:
### Describe the feature

Hi,

is a port to FreeRTOS (using its own TCP/IP stack or lwIP) on the roadmap?
I am thinking of using zenoh-pico in a STM32Cube project.

Thank you!

cguimaraes, 2022-10-27:
Hi @philhoeb ,
We always look for increase the list of supported platforms/frameworks.
We are currently focused on other core aspects of Zenoh, but if there are requests for a specific port we can bring them up in the queue and tackle them in a best-effort way.

Still, we would like to better understand what are your requirements:
  - multi-thread or single-thread support?
  - which would be the initial communication links (e.g., TCP, UDP, Serial, Bluetooth, etc)?
  - would you require scouting capabilities?
  - would you require both unicast and multicast transports?

This information will help us understand what to prioritize on the port, so that you can have access to an initial version that fulfill your needs.
In any case, if you have plans to port it, as an open-source project we appreciate contributions from the community.

philhoeb, 2022-10-31:
Hi @cguimaraes,
Thank you for your response!
The requirements would be: Multi-thread support and communication via TCP.
No need for scouting or multicast transports.

bjsowa, 2023-08-11:
I started writing a FreeRTOS-Plus-TCP port in #239. For now only UDP Unicast transport is supported and no multi-thread support.

Mallets, 2023-10-30:
https://github.com/eclipse-zenoh/zenoh-pico/pull/239 has been merged adding unicast support (TCP/IP) for FreeRTOS. @philhoeb since you stated you do not need for scouting nor multicast transports, I'm closing this issue as resolved. Feel free to reopen it in case.

closed 2023-10-30 by Mallets

philhoeb, 2023-10-30:
Great work, thank you!