eclipse-zenoh/zenoh-pico #145: [Bug] User have to deallocate memory allocated by zenoh-pico

cguimaraes, 2023-01-20:
### Describe the bug

The function `z_keyexpr_to_string` / `zp_keyexpr_resolve` requires user to drop data (`char *`) allocated by it.

This is definitely wrong for Windows platform (memory allocation should not cross DLL boundary) and in certain situations may be unsafe on Unix platform (if caller or callee is compiled with custom malloc/free for example). See https://stackoverflow.com/questions/64574662/allocating-and-deallocating-memory-across-shared-lib-boundaries

Related to https://github.com/eclipse-zenoh/zenoh-c/issues/91 .
Also https://github.com/eclipse-zenoh/zenoh-c/pull/134 .

### To reproduce

N/A

### System info

N/A

philhoeb, 2023-01-20:
As long as the user has to free memory allocated by z_keyexpr_to_string(), the portable z_free() should be used  instead of free() in the examples. In the z_sub.c Zephyr example free() is used to free memory allocated by k_malloc().

cguimaraes, 2023-01-21:
Exactly @philhoeb .
We want to fix its misusing in the examples, but also to have a more explicitly way to identify what has been Zenoh-allocated.
We are already working on it and aligning the fix with Zenoh-C and Zenoh-CPP.

closed 2023-01-23 by cguimaraes