eclipse-zenoh/zenoh-pico #154: Failed to build static library [Bug]

milyin, 2023-01-25:
### Describe the bug

When cmake option BUILD_SHARED_LIBS is set to OFF, build fails with diagnostics:
```
[ 97%] Linking C executable tests/z_msgcodec_test
/usr/bin/ld: lib/libzenohpico.a(tx.c.o): in function `_z_frame_header':
tx.c:(.text+0xb0): multiple definition of `_z_frame_header'; CMakeFiles/z_msgcodec_test.dir/tests/z_msgcodec_test.c.o:z_msgcodec_test.c:(.text+0xb0f0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/z_msgcodec_test.dir/build.make:98: tests/z_msgcodec_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:505: CMakeFiles/z_msgcodec_test.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

### To reproduce

```
git clone https://github.com/eclipse-zenoh/zenoh-pico
cd zenoh-pico && mkdir build && cd build
cmake .. -DBUILD_SHARED_LIBS=OFF
cmake --build .
```

### System info

milyin@GVAARANZ:~/FOO/zenoh-pico/build$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal

milyin@GVAARANZ:~/FOO/zenoh-pico/build$ uname -a
Linux GVAARANZ 5.10.16.3-microsoft-standard-WSL2 #1 SMP Fri Apr 2 22:23:49 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

cguimaraes, 2023-01-26:
Fixed in the PR above.
@milyin if you can give it a quick review.

milyin, 2023-01-26:
> Fixed in the PR above. @milyin if you can give it a quick review.
Looks good, compilation passed

PR #155 (2023-01-27): Failed to build static library (#154)
Fixes #154
closed 2023-01-27 by cguimaraes