eclipse-zenoh/zenoh-pico #157: Align default socket options with Zenoh Rust

cguimaraes, 2023-01-31:
### Describe the feature

> Is your feature request related to a problem?

This is related to a thread discussion going on Discord channel: https://discord.com/channels/914168414178779197/1068452178336940092

> Describe the solution you'd like

The solution is planned to be two-fold:
  1) To have the default values aligned between Zenoh-Pico and Zenoh Rust
  2) To define new `config` properties in the Zenoh `locator` that allows user to change default values.

> Describe alternatives you've considered

N/A

> Additional context

There are advantages and disadvantages depending on the use case, on whether a given socket options is set or not. Due to that, the solution should not rely on static values that cannot be easily changed.
The example given in the Discord channel is related to TCP_NODELAY. Setting this socket options will basically disable the TCP Nagle's algorithm, thus having an impact on the latency.

p-avital, 2023-01-31:
As mentioned on Discord, we should try to avoid offering options that would conflict with our usage. For example, Zenoh-Rust already implements batching, with the same goal as TCP's Nagle, so allowing to enable Nagle on Zenoh-Rust would only needlessly increase delay.

cguimaraes, 2023-01-31:
I don't see a problem on aligning options and allowing users to explicitly decide if they want to change it.
It is their own responsibility if they decide to have both Zenoh batching and TCP's Nagle (in this concrete example).

Users might want to tune their TCP stack for whatever reason and we should not restrict that but, instead, to facilitate it. For example, users might want to disable delayed TCP acknowledgements (with `TCP_QUICKACK`). We cannot pre-define all these parameters.

They are different layers, and my take is that we should keep them independent.
Otherwise, I might argue that if you are on a TCP transport, you might not need to have Zenoh KeepAlive mechanisms.
But this is a whole different discussion...