eclipse-zenoh/zenoh-pico #167: [Bug] Memory corruption when short-enums are enabled.

PetervdPerk-NXP, 2023-03-01:
### Describe the bug

On ARM32 arm-none-eabi-gcc by default is short-enums.
[On some targets, -fshort-enums is the default; this is determined by the ABI.](https://gcc.gnu.org/onlinedocs/gcc-8.2.0/gcc/Structures-unions-enumerations-and-bit-fields-implementation.html#Structures-unions-enumerations-and-bit-fields-implementation)

However the `_z_enum_decode` always assumes it's 4 bytes wide. Hence when short-enums is enabled it will write always 4 bytes to the pointer instead 1-4 (depending on size).

A workaround is to enable the `-fno-short-enums` compile flag which always enforce `int*` size enums.

### To reproduce

Start a subscriber.
When receiving a _Z_MID_JOIN.
Step through `_z_join_decode_na` function.
`msg->_version` gets corrupted when `_z_enum_decode((int *)&msg->_whatami, zbf);` get executed.

### System info

NuttX
ARM32 platform

cguimaraes, 2023-03-01:
I was thinking when I was get bitten by this. It might be fixed in #169 .
In fact, this popped up today in a discussion with CycloneDDS team in a non-related topic.

@PetervdPerk-NXP can you let us know if it fixes the issue?

PetervdPerk-NXP, 2023-03-01:
I've encountered it on Cyclonedds as well indeed.
https://github.com/eclipse-cyclonedds/cyclonedds/issues/1141

So luckily debugging didn't took to long this time.
I will try to test your changes on Friday.

PetervdPerk-NXP, 2023-03-06:
Just tested it on my target #169 fixes this issue

cguimaraes, 2023-03-06:
Thanks for the validation @PetervdPerk-NXP .

closed 2023-03-06 by cguimaraes