eclipse-zenoh/zenoh-pico #168: Build zenoh-pico on Windows 10

Mallets, 2023-03-01:


cguimaraes, 2023-03-01:
The warning is an issue on Windows SDK. If you just want to silence it try `#pragma warning(suppress : 5105)` (https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/c5105).

On the other issue, I am pretty sure the Windows firewall is blocking the multicast traffic (used to scout for a router or peer) in the session establishment.
In order to properly debug it, please provide:
   1) More details about your experimental setup. For example, which entities are running, if all in the same machine or across different machines, and their operating system. The more information you can share with us the better.
   2) Enable ZENOH_DEBUG=3 and share the logs with us.
   3) Wireshark captures are also helpful.

Also, try to execute the examples with a static locator (e.g. `z_pub.exe -e tcp/127.0.0.1:7447`) and let us know what is the result.

lanner17, 2023-03-02:
Hi @cguimaraes, thank you, I resolved it. I upgraded zenohd to v0.7.0-rc from 6. After that I was able to run the pub and sub examples. I should have upgraded first before I raised the question.

cguimaraes, 2023-03-02:
The wire protocol version was bumped between in v0.7.0 as we did some enhancements. For the moment, we require that components to be running in the same version.

As for the new issue, I will double check when I get access to a Windows machine.
In the meantime, can you share a Wireshark capture of those invalid messages?

lanner17, 2023-03-03:
I tried to do a Wireshark capture, but discovered it is not straightforward to do a capture on the local (127.0.0.1) interface on Windows. There are some workarounds, but I am not ready to apply those to the company computer yet. For me, it is not a big deal, I am just exploring zenoh-pico.

lanner17, 2023-03-03:
Hi @cguimaraes, I added an additional interface to the command line and I believe I was able to capture two connection attempts by z_pub that failed. The Wireshark capture is attached. Hope it helps.
[z_pub.zip](https://github.com/eclipse-zenoh/zenoh-pico/files/10884498/z_pub.zip)

cguimaraes, 2023-03-03:
I only see the multicast packets sent by Zenoh for the scouting process.
I do not see any other message related to Zenoh protocol.

Is `zenohd` running and accessible through the multicast path?
Did you make sure that your Windows firewall is not blocking multicast?

lanner17, 2023-04-11:
Hi @cguimaraes , my apologies. I did not have time to dig deeper into this and was just happy to build it on Windows and run it. This is a project on the side to consider alternative middleware and the actual target is actually vxWorks. So, the issue about the multiple connection attempts is not really a concern for me.

cguimaraes, 2023-04-11:
I am closing this issue for the moment then.
If you cross again with it, just reopen it or create a new one issue.

Regarding VxWorks, if that is the actual target we would be happy to know more about what you might need for such port and what is your target use case(s). I suggest either you create a new feature request (in the same tab as issues) or to initiate a discussion on our Discord channel.

closed 2023-04-11 by cguimaraes