eclipse-zenoh/zenoh-pico #173: z_keyexpr_equals API differs from zenoh-c [Bug]

milyin, 2023-03-14:
### Describe the bug

In zenoh-pico `z_keyexpr_equals` returns 0 in case of success: 
https://github.com/eclipse-zenoh/zenoh-pico/blob/5b4734114fad6f7fc484322b53f0f02872171ac5/include/zenoh-pico/api/primitives.h#L212

In zenoh-c the success return value is 1

https://github.com/eclipse-zenoh/zenoh-c/blob/master/include/zenoh_commons.h#L1118

Supposedly there are more API differences like this. The common solution should be found

### To reproduce

-

### System info

-

p-avital, 2023-03-14:
`zenoh-c` is the one that's wrong here, would you mind patching it to follow pico?

It's kinda counter intuitive for `0` to be "equals" in a function that looks like a predicate for equality, but it's to allow error reporting and avoid treating errors as equality.

milyin, 2023-03-14:
I agree, but this also implies that return value should be bool, so the fix should be made in zenoh-c too. The int8_t return value is supposed to be used in sitiations where 0 is success and another value is error code.

p-avital, 2023-03-14:
you're correct, only zenoh-c needs fixing :)

milyin, 2023-03-14:
So what;s the decision: int8_t and 0 for success or bool and true for success for z_keyexpr_vals and z_keyexpr_intersects? I'd prefer bool, but it's up to you to decide

cguimaraes, 2023-03-15:
I am also inline with the latest agreement. We already open the space for more detailed error codes in the future.
Still, we need to remember to explicit mention this in the changelog and documentation, as it might impose changes in user code.

milyin, 2023-03-15:
The API change necessary is not clear yet. The zenohc makes the differences between false return and error return:
```
#[allow(clippy::missing_safety_doc)]
#[no_mangle]
/// Returns ``1`` if `left` and `right` define sets that have at least one key in common, ``0`` if they don't.
/// Returns negative values in case of error (if one of the key expressions is in an invalid state).
pub extern "C" fn z_keyexpr_intersects(left: z_keyexpr_t, right: z_keyexpr_t) -> i8 {
    match (&*left, &*right) {
        (Some(l), Some(r)) => l.intersects(r) as i8,
        _ => i8::MIN,
    }
}
```

The zenoh-pico always returns negative value, no matter is it the error case or normal negative result:
```
int8_t z_keyexpr_intersects(z_keyexpr_t l, z_keyexpr_t r) {
    int8_t ret = 0;

    if ((l._id != Z_RESOURCE_ID_NONE) || (r._id != Z_RESOURCE_ID_NONE)) {
        ret = -1;
    }

    if (_z_keyexpr_intersects(l._suffix, strlen(l._suffix), r._suffix, strlen(r._suffix)) == false) {
        ret = -1;
    }

    return ret;
}
```

Here is some discussion about this topic: https://stackoverflow.com/questions/12728135/general-rule-negative-or-positive-values-for-error-code-in-c-c

So there are multiple conventions an no one is considered as some universal standard.

If we want to both return error values and return different types of success values, we may follow one of these two conventions:

1) negative value - error code 
    zero - success or true
    positive value - false or other kind of success result, like count or some enum value

2) negative value - error code 
    zero - success or false
    positive value - true or other kind of success result, like count or some enum value
 
Both approaches have its advantages and disadvantages. The first one is more universal but counter-intuitive (0 == true). But C programmers used to it. Second one is more natural but may cause unexpected errors when user just treats return value as boolean and therefore erroneously treats error code as success.

For following any of these approaches both zenoh-c and zenoh-pico should be updated

I'd vote to approach (1) - as more consistent one

milyin, 2023-03-16:
Made pull requests to API change accordingly to variant (1). For zenoh-cpp development I need to have zenohpico and zenohcpp APIs aligned, so I'm using these updates as temporary approach for my current development

cguimaraes, 2023-03-21:
After discussion between @cguimaraes @milyin @p-avital we decided with the following approach as it imposes a minimal break on the API:

```
0 --> true
-1 --> false
other negative values --> errors
```

Still, this point might be revisited in the future when Zenoh has a proper definition of error codes.

closed 2023-03-22 by cguimaraes