eclipse-zenoh/zenoh-pico #193: [Improvement] Messages are by design 65535 bytes maximum (2-bytes)

cguimaraes, 2023-04-05:
### Describe the bug

`size_t` is being used to handle the size of Zenoh messages.
Since Zenoh messages cannot longer than 65535 bytes, its maximum length can be represented in a `uint16_t`.

### To reproduce

N/A

### System info

N/A

jean-roland, 2023-12-27:
The issue with truncated fixed length types such as `uint16_t`, `uint8_t`... is that, depending on processor architecture, using them in local variables or function parameters (where registers will typically be used) will force the compiler to emit masking instructions.

So they really should only be used in struct to reduce memory size, or if there is an imperative to limit the value and the masking is necessary. Otherwise, we should use the `uint_fast16_t`, `uint_fast8_t`...

 If we go this route, then I suggest this issue should be included in a more broader scope issue.

cguimaraes, 2023-12-28:
`uint_fast_*_t` or `uint_least*_t` were in the backlog to be considered, but there was a first check on how does it impact the MISRA-C compliance.

This check is still pending and, in my opinion, it shall be done before moving into those types.