eclipse-zenoh/zenoh-pico #204: Protocol check is ignored causing issues on embedded [Bug]

hairymnstr, 2023-04-14:
### Describe the bug

There's an if statement checking the protocol part of the locator matches but there's nothing preventing the next section of code running if this check fails.  If I have a locator "serial/serial0" then the test at https://github.com/eclipse-zenoh/zenoh-pico/blob/b77b03749d3156d2187b00bd8212ee3347271f0f/src/link/unicast/tcp.c#L78 will fail and set an error in `ret` but the code then goes on to try and analyse this locator as a TCP locator which causes an issue since the string doesn't have a ":" character that the TCP locator expects.

### To reproduce

Run a demo in valgrind, using a locator string of "serial//dev/ttyUSB0" in one of the demos e.g.:

valgrind ./z_put -e serial//dev/ttyUSB0

the output of valgrind indicates this affects multiple transports (although on Linux it seems to run successfully):

```==32898== Memcheck, a memory error detector
==32898== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==32898== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==32898== Command: ./z_put -e serial//dev/ttyUSB0
==32898== 
Opening session...
==32898== Argument 'size' of function malloc has a fishy (possibly negative) value: -78143311
==32898==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==32898==    by 0x120593: z_malloc (system.c:95)
==32898==    by 0x122AAD: __z_parse_address_segment_tcp (tcp.c:66)
==32898==    by 0x122B1E: _z_endpoint_tcp_valid (tcp.c:82)
==32898==    by 0x121DF7: _z_open_link (link.c:32)
==32898==    by 0x11C205: _z_new_transport_client (manager.c:26)
==32898==    by 0x11C51F: _z_new_transport (manager.c:102)
==32898==    by 0x1118D1: __z_open_inner (session.c:33)
==32898==    by 0x111B58: _z_open (session.c:109)
==32898==    by 0x10CBDB: z_open (api.c:516)
==32898==    by 0x10AAB4: main (z_put.c:62)
==32898== 
==32898== Argument 'size' of function malloc has a fishy (possibly negative) value: -78143311
==32898==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==32898==    by 0x120593: z_malloc (system.c:95)
==32898==    by 0x123089: __z_parse_address_segment_udp_unicast (udp.c:66)
==32898==    by 0x1230FA: _z_endpoint_udp_unicast_valid (udp.c:82)
==32898==    by 0x121E1F: _z_open_link (link.c:37)
==32898==    by 0x11C205: _z_new_transport_client (manager.c:26)
==32898==    by 0x11C51F: _z_new_transport (manager.c:102)
==32898==    by 0x1118D1: __z_open_inner (session.c:33)
==32898==    by 0x111B58: _z_open (session.c:109)
==32898==    by 0x10CBDB: z_open (api.c:516)
==32898==    by 0x10AAB4: main (z_put.c:62)
==32898== 
Unable to open session!
==32898== 
==32898== HEAP SUMMARY:
==32898==     in use at exit: 0 bytes in 0 blocks
==32898==   total heap usage: 26 allocs, 26 frees, 2,177 bytes allocated
==32898== 
==32898== All heap blocks were freed -- no leaks are possible
==32898== 
==32898== For lists of detected and suppressed errors, rerun with: -s
==32898== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)```

On my embedded platform the negative malloc() call triggers an unhelpful assert just saying out of memory.

### System info

version tested: b77b03749d3156d2187b00bd8212ee3347271f0f
happens on embedded and Linux platforms.

cguimaraes, 2023-04-20:
@hairymnstr can you validate PR #205 please?

hairymnstr, 2023-04-20:
Looks good to me.

closed 2023-04-20 by cguimaraes