eclipse-zenoh/zenoh-pico #215: Expose timeout option in z_get_options_t

cguimaraes, 2023-05-16:
### Describe the feature

Describe the feature

The timeout option (from rust: https://docs.rs/zenoh/0.7.0-rc/zenoh/query/struct.GetBuilder.html#method.timeout) should be exposed in zenoh-pico get options `z_get_options_t`.

oteffahi, 2024-02-20:
The absence of a timeout option in zenoh-pico seems to cause a bug where an instance can hang forever if it performs a `get` and at least one queryable that received it never responds. Furthermore, the zenoh-pico instance hangs forever even if said queryable is exited.

Steps to reproduce (using `0.10.1-rc`):
- Modify any `z_queryable` example to insert a `sleep` call between the reception of the `get` and sending the reponse. In my tests I use `zenoh-python/z_queryable.py`.
- Run the `z_queryable` example.
- Run the `zenoh-pico/z_get` example: `zenoh-pico/z_get` never times-out during the queryable's sleep call.
- Stop the `z_queryable` instance: `zenoh-pico/z_get` hangs forever.

sashacmc, 2024-03-15:
@milyin please assign it to me

PR #375 (2024-03-21): Expose timeout option in z_get_options_t
Closes #215
closed 2024-03-21 by Mallets