eclipse-zenoh/zenoh-pico #229: [Bug] Build breaks on Zephyr RTOS v3.4.0

entire, 2023-07-06:
### Describe the bug

### Describe the bug

Seems like there was a change that breaks `zenoh-pico/src/system/zephyr/network.c` in the network interface APIs, since Zephyr v3.4.0 it now take additional, struct net_if * iface parameter:

[:c:func:`net_if_ipv4_maddr_join`](https://github.com/zephyrproject-rtos/zephyr/blob/6e27b7d56622207870944de78dbee394421b5f16/doc/releases/release-notes-3.4.rst#id105)
[:c:func:`net_if_ipv4_maddr_leave`](https://github.com/zephyrproject-rtos/zephyr/blob/6e27b7d56622207870944de78dbee394421b5f16/doc/releases/release-notes-3.4.rst#id107)
[:c:func:`net_if_ipv6_maddr_join`](https://github.com/zephyrproject-rtos/zephyr/blob/6e27b7d56622207870944de78dbee394421b5f16/doc/releases/release-notes-3.4.rst#id109)
[:c:func:`net_if_ipv6_maddr_leave`](https://github.com/zephyrproject-rtos/zephyr/blob/6e27b7d56622207870944de78dbee394421b5f16/doc/releases/release-notes-3.4.rst#id111)

It works with Zephyr RTOS v3.3.0 before the change was made. 

### To reproduce

### To reproduce
- Fresh build from the https://github.com/eclipse-zenoh/zenoh-pico/commit/9b663d5b8aa0e7aa35f0da9285622e0c8cc25649 commit
- Build with Zephyr RTOS v3.4.0 tag or later
- ran `west init && west build` locally, followed https://github.com/cguimaraes/zenoh-pico-zephyr-example example

### System info

- MacOS 13.3.1 Ventura (Mac M1)
- Zephyr RTOS v3.4.0
- Zenoh-pico 0.7.2

cguimaraes, 2023-07-06:
Thanks for reporting it.
At this point, we shall support:
  - Zephyr 2(.7.0+) as the LTS release
  - Zephyr 3.3.0 and below
  - Zephyr 3.4.0 and above

@entire Do you know if there is any Zephyr environment variable that can be used at compile time to identify the version?

bryceschober, 2023-07-06:
@cguimaraes See https://docs.zephyrproject.org/latest/doxygen/html/kernel__version_8h.html

bryceschober, 2023-07-06:
Also in 2.7: v

bryceschober, 2023-07-06:
@cguimaraes It looks like the low-level "interface" is to `#include "versio.h"`, whose contents are generated by (and documented in) the `version.cmake` file ([2.7](https://github.com/zephyrproject-rtos/zephyr/blob/v2.7-auditable-branch/cmake/version.cmake), [3.3-main](https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/version.cmake)).

cguimaraes, 2023-07-12:
Thank you @bryceschober for the input. It was very helpful to improve the Zephyr version detection within Zenoh-Pico.

There is a partial fix for this issue, until the new Zephyr multicast API is integrated: https://github.com/eclipse-zenoh/zenoh-pico/pull/230

@entire if this is a blocking point for you but you do not need Zenoh multicast transport, you might want to set `Z_LINK_UDP_MULTICAST` to `0`.

cguimaraes, 2023-07-12:
#230 contains now a complete fix for it.
Missing proper testing before merging.

@entire could you help us testing it on your setup?

closed 2023-07-26 by cguimaraes

entire, 2023-08-23:
seems to work, thanks! @cguimaraes

PR #230 (2023-07-26): Improve Zephyr version detection
Partially fix #229 .