eclipse-zenoh/zenoh-pico #234: [Bug] Installation fails when building from source

imstevenpmwork, 2023-07-31:
### Describe the bug

After
```
make
make install
```
 I got:
 ```
-- Up-to-date: /usr/local/include/zenoh-pico/session/subscription.h
-- Up-to-date: /usr/local/include/zenoh-pico/session/publication.h
CMake Error at build/cmake_install.cmake:65 (file):
  file cannot create directory: /cmake/zenohpico.  Maybe need administrative
  privileges.
```

### To reproduce

Follow the instructions in the readme for building from source

### System info

OS: macOS 13.4.1 22F82 arm64

Mallets, 2023-07-31:
To install it under `/usr/local` you need administrative privileges. Can you try with `sudo`?

imstevenpmwork, 2023-07-31:
Oh I did, I forgot to mention that.
I did run it with sudo, yea

commit 8abd39bf (2023-09-18): Fix #234 (#250)
closed 2023-09-18 by Mallets