eclipse-zenoh/zenoh-pico #241: Add support for Arduino (atmel based boards)

freol35241, 2023-08-23:
### Describe the feature

Trying to compile for one of the atmel based arduino boards currently error out with "Platform unknown".

cguimaraes, 2023-08-23:
Can you provide more details on how are you compiling your project?
Depending on how you do it, you might need to define additional environment flags.

freol35241, 2023-08-23:
Hi, I am using a platformio setup defined as following (platformio.ini):
```
platform = atmelavr
board = yun
framework = arduino
lib_deps = https://github.com/eclipse-zenoh/zenoh-pico#0.7.0-rc
```

And I am trying to compile the [z_scout.ino](https://github.com/eclipse-zenoh/zenoh-pico/blob/master/examples/arduino/z_scout.ino) example from this repository.

The compile error originates from [here](https://github.com/eclipse-zenoh/zenoh-pico/blob/master/include/zenoh-pico/system/platform.h#L40) which lead me to [`extra_script.py`](https://github.com/eclipse-zenoh/zenoh-pico/blob/master/extra_script.py#L33-L46) which only seems to support the `espressif32` platform for the `arduino` framework.

Perhaps there is just a need for a set of different environment flags for the `atmelavr` platform?

ktetsuo, 2024-08-31:
I am facing the same problem.
Please let me know the current status of this problem.