eclipse-zenoh/zenoh-pico #242: API inconsistency: need function to check if z_keyexpr_t is Z_RESOURCE_ID_NONE

milyin, 2023-08-25:
### Describe the feature

The `z_keyexpr_t` have the following definition:
```C
typedef struct {
    _z_zint_t _id;
    const char *_suffix;
} _z_keyexpr_t;
```
The `z_keyexpr_t` instance is considered valid if it is registered in the `Session` object (let's name it **declared**) or if it have the key expression string value stored inside (let's name it **standalone**):
```C
_Bool z_keyexpr_is_initialized(const z_keyexpr_t *keyexpr) {
    _Bool ret = false;
    if ((keyexpr->_id != Z_RESOURCE_ID_NONE) || (keyexpr->_suffix != NULL)) {
        ret = true;
    }
    return ret;
}
```
But all key expression processing functions, like `z_keyexpr_includes`, `z_keyexpr_intersects`, etc. are able to process only **standalone** instances, otherwise they return error:
```C
int8_t z_keyexpr_includes(z_keyexpr_t l, z_keyexpr_t r) {
    int8_t ret = 0;
    if ((l._id == Z_RESOURCE_ID_NONE) && (r._id == Z_RESOURCE_ID_NONE)) {
        ret = zp_keyexpr_includes_null_terminated(l._suffix, r._suffix);
    } else {
        ret = _Z_ERR_GENERIC;
    }
    return ret;
}
```
There is no direct way to determine through API, is the valid key expression **standalone** or **declared**. It's possible to check it through `z_keyexpr_as_bytes`, but this function performs extra work -  creates `z_bytes_t` structure and calls `strlen`.

I think we need API function like this:
```C
_Bool zp_keyexpr_is_declared(const z_keyexpr_t *keyexpr) {
    _Bool ret = false;
    if (keyexpr->_id != Z_RESOURCE_ID_NONE) {
        ret = true;
    }
    return ret;
}
```
If this function returns `TRUE`, this means that the `z_keyexpr_t` instance is valid and declared in `Session` and it's text value can be extracted with `zp_keyexpr_resolve`. Currently there is no way to determine this.

This functionality is necessary to fix issue https://github.com/eclipse-zenoh/zenoh-cpp/issues/20 in zenoh-cpp

milyin, 2024-01-08:
The issue https://github.com/eclipse-zenoh/zenoh-cpp/issues/20 was fixed in zenoh-cpp without `is_declared`function for now - by forcedly call `zp_keyexpr_resolve`. 
https://github.com/eclipse-zenoh/zenoh-cpp/pull/65/files#diff-43c54af67719e2192160a46532508beba52452a2de558291cee6ae6db650cca2R666-R668
The solution can be made more effective with this `is_declared` function - by avoiding `zp_keyexpr_resolve` for `Z_RESOURCE_ID_NONE` keyexprs, so such function is still useful

jean-roland, 2024-01-09:
Let's take it the other way, if you need to cheaply differentiate between **declared** and **standalone**, couldn't this function works as well?

```C
_Bool zp_keyexpr_is_standalone(const z_keyexpr_t *keyexpr) {
    _Bool ret = false;
    if (keyexpr->_suffix != NULL) {
        ret = true;
    }
    return ret;
}
```

closed 2024-01-16 by milyin