eclipse-zenoh/zenoh-pico #249: [Bug] "unknown platform" building with PlatformIO for the target ESP32

andreaz98, 2023-09-15:
### Describe the bug

When building an application for the ESP32 with PlatformIO I get the error message `unknown platform`. I expect PlatformIO recognizes it has to build the ESP32 path on zenoh-pico but I had to add the flags `build_flags=-DZENOH_ESPIDF` to the file `platformio.ini` for the building process to work properly. Follows the full building's log when the build flags are NOT set:

```
Executing task: platformio run --environment az-delivery-devkit-v4 

Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (6.0.0+sha.98046b6) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50000.0 (5.0.0) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- zenoh-pico @ 0.10.0.0
Building in release mode
Compiling .pio/build/az-delivery-devkit-v4/src/main.o
In file included from lib/zenoh-pico/include/zenoh-pico/collections/element.h:21,
                 from lib/zenoh-pico/include/zenoh-pico/net/session.h:20,
                 from lib/zenoh-pico/include/zenoh-pico/net/publish.h:18,
                 from lib/zenoh-pico/include/zenoh-pico/api/types.h:17,
                 from lib/zenoh-pico/include/zenoh-pico/api/primitives.h:21,
                 from lib/zenoh-pico/include/zenoh-pico/api/macros.h:17,
                 from lib/zenoh-pico/include/zenoh-pico.h:25,
                 from src/main.c:22:
lib/zenoh-pico/include/zenoh-pico/system/platform.h:40:2: error: #error "Unknown platform"
   40 | #error "Unknown platform"
      |  ^~~~~
In file included from lib/zenoh-pico/include/zenoh-pico/link/link.h:24,
                 from lib/zenoh-pico/include/zenoh-pico/link/manager.h:20,
                 from lib/zenoh-pico/include/zenoh-pico/transport/manager.h:19,
                 from lib/zenoh-pico/include/zenoh-pico/session/session.h:28,
                 from lib/zenoh-pico/include/zenoh-pico/net/session.h:24,
                 from lib/zenoh-pico/include/zenoh-pico/net/publish.h:18,
                 from lib/zenoh-pico/include/zenoh-pico/api/types.h:17,
                 from lib/zenoh-pico/include/zenoh-pico/api/primitives.h:21,
                 from lib/zenoh-pico/include/zenoh-pico/api/macros.h:17,
                 from lib/zenoh-pico/include/zenoh-pico.h:25,
                 from src/main.c:22:
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:26:5: error: unknown type name '_z_sys_net_socket_t'
   26 |     _z_sys_net_socket_t _sock;
      |     ^~~~~~~~~~~~~~~~~~~
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:27:5: error: unknown type name '_z_sys_net_endpoint_t'
   27 |     _z_sys_net_endpoint_t _rep;
      |     ^~~~~~~~~~~~~~~~~~~~~
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:30:31: error: unknown type name '_z_sys_net_endpoint_t'
   30 | int8_t _z_create_endpoint_tcp(_z_sys_net_endpoint_t *ep, const char *s_address, const char *s_port);
      |                               ^~~~~~~~~~~~~~~~~~~~~
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:31:27: error: unknown type name '_z_sys_net_endpoint_t'
   31 | void _z_free_endpoint_tcp(_z_sys_net_endpoint_t *ep);
      |                           ^~~~~~~~~~~~~~~~~~~~~
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:33:20: error: unknown type name '_z_sys_net_socket_t'; did you mean '_z_tcp_socket_t'?
   33 | int8_t _z_open_tcp(_z_sys_net_socket_t *sock, const _z_sys_net_endpoint_t rep, uint32_t tout);
      |                    ^~~~~~~~~~~~~~~~~~~
      |                    _z_tcp_socket_t
lib/zenoh-pico/include/zenoh-pico/system/link/tcp.h:33:53: error: unknown type name '_z_sys_net_endpoint_t'
   33 | int8_t _z_open_tcp(_z_sys_net_socket_t *sock, const _z_sys_net_endpoint_t rep, uint32_t tout);
[... 720 lines cut ...]
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = espidf
; build_flags=-DZENOH_ESPIDF
```
2. `git clone https://github.com/andreaz98/zenoh-pico` on the `lib` folder of the PlatformIO project just created.
3. Include `#include "zenoh-pico.h"` in the main of your PlatformIO project.
4. Build the project via the UI of PlatformIO.
5. See the errors above (except the last part which just takes a function renaming to solve the errors).

### System info

- ESP32 espressif32 az-delivery-devkit-v4
- zenoh-pico commit: `6b9e58669bc17dc0b23289d5c97e8a85998d22aa` (refers to the fork `andreaz98/zenoh-pico`)

anhaabaete, 2023-09-28:
here is the same

andreaz98, 2023-09-28:
> here is the same

please, to your `platformio.ini` add the following line: `build_flags = -DZENOH_ESPIDF`. This is a bug of PlatformIO (see https://github.com/eclipse-zenoh/zenoh-pico/blob/master/extra_script.py#L62). Hence, defining ZENOH_ESPIDF must be done manually.

anhaabaete, 2023-09-30:
@andreaz98 Thanks, I add the flag on platformio.ini and the test with code [here](https://zenoh.io/blog/2021-10-04-zenoh-pico-guide/) is don't work. But I tried with the code example from git repository and in this case is well done! So, maybe the code at the guide site example may be buged, no?

But to be clean, with the example from the GIT, it compile very well.

cguimaraes, 2023-10-01:
@anhaabaete note that the code in the blog post is from October 2021.
Since then, Zenoh had a refactor on both the user API (a couple releases ago) and on the wire protocol (released last week).

It would be better if you use the examples provided in the repository itself: https://github.com/eclipse-zenoh/zenoh-pico/tree/master/examples/espidf

anhaabaete, 2023-10-01:
@cguimaraes  (Guimaraes?) Do you speak portuguese?

So... ok, I want update the blog. Is possible?

cguimaraes, 2023-10-01:
> @cguimaraes (Guimaraes?) Do you speak portuguese?

Sim...todos os dias :-)

> So... ok, I want update the blog. Is possible?

As a decision, we opted to not update the blog posts upon every new release. So, they shall be contextualized in time.

Unfortunately, old blog posts are not referring which version was being used. But, if you take corresponding `z_pub` or `z_sub` that we have in the examples folder ( https://github.com/eclipse-zenoh/zenoh-pico/tree/master/examples/espidf ), instead of the one in the blog post, everything shall work as described.

Feel free to text us in Zenoh's Discord channel.

cguimaraes, 2023-10-12:
I am closing this issue, and it relates to an external tool.

closed 2023-10-12 by cguimaraes