eclipse-zenoh/zenoh-pico #259: [Bug] z_sub example leaks allocated suffix in z_declare_subscriber

hwoithe, 2023-10-10:
### Describe the bug

The example z_sub.c calls [`z_declare_subscriber()`](https://github.com/eclipse-zenoh/zenoh-pico/blob/c4f78b2165741578a259a72eab140afa5233303b/examples/unix/c11/z_sub.c#L87) with the keyexpr "demo/example/**".
This causes the `z_declare_subscriber()` to [allocate memory](https://github.com/eclipse-zenoh/zenoh-pico/blob/c4f78b2165741578a259a72eab140afa5233303b/src/api/api.c#L827) for a suffix that is never freed before the program ends.
Calling `_z_keyexpr_clear(&keyexpr)` after `_z_declare_resource()` removes the leak but I am unsure if this is the correct solution as I am not familiar with the zenoh-pico codebase.

### To reproduce

1. Console 1: zenohd -l tcp/localhost:7447 --no-multicast-scouting
2. Console 2: build/examples/z_sub -m client -e tcp/127.0.0.1:7447
3. Console 2: Press 'q' to quit

### System info

Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:    22.04
Codename:   jammy

CPU: Intel(R) Core(TM) i9-7920X CPU @ 2.90GHz

zenoh commit: 1f935b16baaaf45c0c18c8fafd0d03b774f4b4ca
zenoh-pico commit: c4f78b2165741578a259a72eab140afa5233303b

cguimaraes, 2023-10-10:
Hi @hwoithe,
If you have a fix for this, feel free to contribute with a PR.
We will make sure to review it and merge it.
Thanks in advance.

hwoithe, 2023-10-12:
Thanks @cguimaraes for the quick response. I will continue to investigate whether my suggested fix is correct. Meanwhile, I am working through the Eclipse Contributor Agreement with my employer. I am unsure how long this will take. If there's a need to push this through rather quickly, it may be faster for an already approved committer to perform the fix.

cguimaraes, 2023-10-12:
It seems that @p-avital provided a fix.
@hwoithe can you test it on your scenario please?

hwoithe, 2023-10-12:
Awesome, thank you! I tested the branch and it no longer seems to leak memory.

PR #262 (2023-10-18): fix memory leak in z_declare_subscriber
fixes #259
closed 2023-10-18 by p-avital