eclipse-zenoh/zenoh-pico #270: [Bug] Use `CLOCK_MONOTONIC` where available instead of `CLOCK_REALTIME`

bryceschober, 2023-10-25:
### Describe the bug

The PR #269 brought my attention to the fact that `z_clock_now()` uses `clock_gettime(CLOCK_REALTIME...)` on most posix-like platforms. But [POSIX specifies](https://pubs.opengroup.org/onlinepubs/000095399/basedefs/time.h.html) that `CLOCK_MONOTONIC` is the only one appropriate for usage as a relative timing reference, because it will never go backwards as a result of calling [`clock_settime()`](https://pubs.opengroup.org/onlinepubs/000095399/functions/clock_getres.html). In practice, this is [definitely true on Linux systems](https://stackoverflow.com/a/3527632), and is also true in [the Zephyr implementation](https://github.com/zephyrproject-rtos/zephyr/blob/master/lib/posix/clock.c#L28). The [esp-idf implementation](https://github.com/espressif/esp-idf/blob/master/components/newlib/time.c#L250) looks like it has the same properties.

So in practice, relative timing (the most common usage of time in embedded systems at the library / protocol level) should only ever use `CLOCK_MONOTONIC`, and `CLOCK_REALTIME` should only ever be used when the use case is actually user-level wall-clock time.

### To reproduce

Sorry, but I haven't tried this out in actual practice... but it's pretty widely-known best practices for embedded posix-like systems.

1. Start a long-running example.
2. Use some means of setting the system wall clock backwards that ultimately uses `clock_settime()`, in simulation of an NTP update, etc.
3. What happens? :shrug:  Depending on the system implementation, protocol timing could be very broken...

### System info

Any posix-like system using `CLOCK_REALTIME`.

cguimaraes, 2023-10-26:
Indeed. 

In fact, we have both `z_clock_*` and `z_time_*` primitives which original purpose was to include support for both time computations.

But I am now wondering if for this case, we want the `CLOCK_MONOTONIC` or the `CLOCK_MONOTONIC_RAW`, since the former might still have its clock oscillator disciplined by NTP.

cguimaraes, 2023-10-26:
Being addressed by @jean-roland in #269

Mallets, 2023-10-26:
> Indeed.
>
> In fact, we have both `z_clock_*` and `z_time_*` primitives which original purpose was to include support for both time computations.
>
> But I am now wondering if for this case, we want the [...]

I've just checked what Rust does: https://doc.rust-lang.org/std/time/struct.Instant.html#underlying-system-calls
`CLOCK_MONOTONIC` is used for Unix platforms. For compatibility reasons, I'd then prefer to keep `CLOCK_MONOTONIC`. By doing so,  we can have the same behaviour in Zenoh-Pico and Zenoh-Rust-based API for what concerns time counting.

Mallets, 2023-10-26:
Closed by #269

closed 2023-10-26 by Mallets