eclipse-zenoh/zenoh-pico #278: failed to use `find_package(zenohpico)` [Bug]

milyin, 2023-11-09:
### Describe the bug

There are 3 proper ways to use zenoh-pico library in customer cmake project:
- `find_package(zenohpico)`
- `add_subdirectory(../zenoh-pico)`
- `FetchContent` from github repository, which is just a wrapper over the `add_subdirectory` command

The `find_package` includes file `lib/cmake/zenohpico/zenohpicoConfig.cmake` generated from https://github.com/eclipse-zenoh/zenoh-pico/blob/master/PackageConfig.cmake.in 

The `add_subdirectory` includes project's CMakeLists.txt: https://github.com/eclipse-zenoh/zenoh-pico/blob/master/CMakeLists.txt

This approach works well for zenoh-c and zenoh-cpp. See for example testing of these featues in zenoh-c CI:

Install zenohc package:
https://github.com/eclipse-zenoh/zenoh-c/blob/master/.github/workflows/ci.yml#L43-L45
Test building examples using zenoh-c directory as subproject:
https://github.com/eclipse-zenoh/zenoh-c/blob/master/.github/workflows/ci.yml#L72-L74
Test building examples using installed zenohc package:
https://github.com/eclipse-zenoh/zenoh-c/blob/master/.github/workflows/ci.yml#L79-L81

The CMakeLists.txt in `examples` directory of zenoh-pico also allows to test both approaches. At this moment building with `find_package` doesnt work for zenoh-pico

It's necessary to fix it and add testing of both approaches to CI, like it's done in zenoh-c

### To reproduce

Install zenohpico as a package locally
```
git clone git@github.com:eclipse-zenoh/zenoh-pico.git
cmake -S zenoh-pico -B build -DCMAKE_INSTALL_PREFIX=~/FOO/install
cmake --build build --target install
```

Build examples using zenoh-pico as packaget:
```
cmake -S zenoh-pico/examples -B build_examples
cmake -S zenoh-pico/examples -B build_examples -DZENOHPICO_SOURCE=PACKAGE
```

This fails due to misconfigured line
https://github.com/eclipse-zenoh/zenoh-pico/blob/master/examples/CMakeLists.txt#L7
(zenohc instead of zenohpico::static)

But even after fix this the more problem comes: include direcrories and preprocessor settings are not provided to dependent project correctly

### System info

Ubuntu 22.04

Mallets, 2024-01-19:
@milyin @jean-roland is this issue still open?

jean-roland, 2024-01-19:
Yeah, this haven't been fixed yet.

milyin, 2024-01-25:
When reviewing fix for bug, it was found that zenop-pico package installation is not correct now, this is not only single typo in example.

First, I'll describe the whole concept and how package installation works:
1. package is searched by cmake's command find_package by it's name: "zenohc" or "zenohpico". 
2. techninally, package is a file "/lib/cmake/zenohc/zenohcConfig.cmake" which is included into user's CMakeLists.txt. This file is generated from template https://github.com/eclipse-zenoh/zenoh-c/blob/main/install/PackageConfig.cmake.in
3. inside the file the lbrary targets are declared.  Zenoh-c declares 3 targets: zenohc::static, zenohc::shared and zenohc::lib which is shared or static, depending on ZENOHC_LIB_STATIC setting
4. each target is configured with set of properties, some of them specific to target. These properties are:
```CMake
# path to library file
set_property(TARGET __zenohc_static PROPERTY IMPORTED_LOCATION "/home/milyin/FOO/install/lib/libzenohc.a")
# additional libraries to be linked when linking this library
target_link_libraries(__zenohc_static INTERFACE rt;pthread;m;dl)
# include directories for the library
target_include_directories(__zenohc_static INTERFACE "/home/milyin/FOO/install/include")
# say that shared library doesn't contain soname property: https://en.wikipedia.org/wiki/Soname
set_target_properties(__zenohc_shared PROPERTIES IMPORTED_NO_SONAME TRUE)
# define additional preprocessor settings
target_compile_definitions(__zenohc_shared INTERFACE ZENOHC_DYN_LIB)
```
5. on the user's side it's enough to include the package and add dependency on necessary library. No additional settings are required:
```CMake
find_package(zenohc)
add_executable(app)
add_dependencies(app zenohc::lib)
target_link_libraries(app PRIVATE zenohc::lib)
```

For zenoh-pico there are several problems:
- `IMPORTED_LOCATION` is not set to correct path of the library file
- `target_compile_definitions` is not set (`ZENOH_LINUX` or `ZENOH_WINDOWS`, depending on platform, defintion for shared library import should be here also, I suppose)
- the library target names in zenoh-pico main CMakeLists.txt are not in accord with names in package: it's just `zenohpico` in CMakeLists.txt, but `zenohpico::static` and `zenohpico::lib` in package. This doesn't alllow to interchange `find_package` / `add_subdirectory`
...may be more of it

The main problem with zenoh-pico is that it don't use target-local properties (i.e. it uses global `add_definitions` instead of `target_compile_definitions` for example, same thing for paths, etc.). There is a hack to make zenoh-pico work with zenoh-cpp build: https://github.com/eclipse-zenoh/zenoh-pico/blob/main/CMakeLists.txt#L48-L51. But using global properties in package config file is definitely wrong.
So the choice is: 
1. refactor zenoh-pico CMakeLists.txt, getting rid of global properties
2. only implement correct  https://github.com/eclipse-zenoh/zenoh-pico/blob/main/PackageConfig.cmake.in with all necessary target properties, keep main CMakeLists with global properties for now

In the second case it will be hard to keep PackageConfig and CmakeLists.txt in sync, as they will be significantly different. Compare with how it's made in zenoh-cpp, where these files are practically the same:
https://github.com/eclipse-zenoh/zenoh-cpp/blob/main/CMakeLists.txt
https://github.com/eclipse-zenoh/zenoh-cpp/blob/main/install/PackageConfig.cmake.in

closed 2024-01-26 by Mallets