eclipse-zenoh/zenoh-pico #282: Segmentation fault occurs when receiving messages while undeclaring subscriber [Bug]

gmartin82, 2023-11-24:
### Describe the bug

A race condition exists in Zenoh Pico which can cause a segmentation fault to occur if messages are received at the same time as calling `z_undeclare_subscriber()`. This is more evident where messages are received at a high rate.

The issue has been reproduced in the `z_sub` example with the following stack trace in gdb from the dumped core:

`#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
#1  0x00007f47bf20cc1d in __z_get_subscriptions_by_key ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#2  0x00007f47bf20cfa3 in _z_trigger_subscriptions ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#3  0x00007f47bf20a7a4 in _z_trigger_push ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#4  0x00007f47bf20c219 in _z_handle_network_message ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#5  0x00007f47bf210174 in _z_unicast_handle_transport_message ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#6  0x00007f47bf20fcec in _zp_unicast_read_task ()
   from /home/geoff/Development/git/zenoh-pico/build/lib/libzenohpico.so
#7  0x00007f47bf025ac3 in start_thread (arg=<optimised out>)
    at ./nptl/pthread_create.c:442
#8  0x00007f47bf0b7a40 in clone3 ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81`

The segfault appears to occur on [this](https://github.com/eclipse-zenoh/zenoh-pico/blob/b5483d79cb857d2d5c45fd6585d0d1c493991f3a/src/session/subscription.c#L65) line due to `sub->ptr->_key.suffix` being null.

The issue may be cause by [this](https://github.com/eclipse-zenoh/zenoh-pico/blob/b5483d79cb857d2d5c45fd6585d0d1c493991f3a/src/net/primitives.c#L228) line with the key being moved and stolen in the call.

### To reproduce

1. Adapt the `z_pub` example to send messages as quickly as possible and continuously.
2. Compile the `z_pub` and `z_sub` examples.
3. Run the `z_pub` example.
4. Run the `z_sub` example.
5. Quit the `z_pub` example by pressing `q` then `return`. Note several runs may be required to reproduce the issue.

### System info

- Platform: Ubuntu 22.04 64-bit
- CPU: Intel i7-1260P
- Zenoh version: master (b5483d79cb857d2d5c45fd6585d0d1c493991f3a)

jean-roland, 2023-11-27:
Will look into it but it might be interesting to add a CI stress test to catch these kind of issues earlier.

jean-roland, 2023-12-06:
So yeah, the undeclare subscriber process is:
1) steal the keyexpr
2) send the declare n message
3) remove subscription from session

PR #305 (2024-01-08): Recopy key instead of steal when undeclaring
This fixes #282, the undeclare process for queryable and subscription was:

    1) steal the keyexpr
    2) send the declare N message
    3) remove subscription/queryable from session
    
The delay between 1) and 3) was the source of the problem and stealing the keyepr was probably useful before the introduction of refcounting for subscriptions and queryables, but it is no longer necessary.
closed 2024-01-08 by Mallets