eclipse-zenoh/zenoh-pico #285: [Bug] Zephyr CI is broken

jean-roland, 2023-11-30:
### Describe the bug

The Zephyr workflow seems to be broken for some reason:

```
Compiling .pio/build/nucleo_f767zi/kernel/zephyr/kernel/kheap.c.o
/home/runner/.platformio/packages/framework-zephyr/lib/posix/signal.c:6:10: fatal error: posix/strsignal_table.h: No such file or directory
    6 | #include "posix/strsignal_table.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/nucleo_f767zi/lib__posix/zephyr/lib/posix/signal.c.o] Error 1
========================= [FAILED] Took 25.62 seconds =========================
Error: Process completed with exit code 1.
```

### To reproduce

Run the Zephyr workflow

### System info

This is a CI thing

cguimaraes, 2023-12-01:
There was an update on the Zephyr package in the last days.
It might be that the Zenoh-Pico port needs to be updated due to that.

jean-roland, 2023-12-04:
Platformio updated their Zephyr package to Zephyr 3.5 and it seems there was an issue for stm32, they fixed it. (https://community.platformio.org/t/issues-building-on-zephyr/37069/4)

Have some changes to push on our side regarding this update, will make a PR and closing this issue.

closed 2023-12-04 by jean-roland