eclipse-zenoh/zenoh-pico #288: [Bug] Memory usage continues to increase when sending data in peer mode

gmartin82, 2023-12-05:
### Describe the bug

While publishing data in peer mode via multicast I noticed that memory usage (as shown by `top`) continued to increase over time indicating a possible memory leak. When sending large volumes of data frequently (as is the case when running a performance test for example) memory usage can increase substantially and can end up consuming all available system memory.

I have been able to reproduce the leak with the ```z_pub``` example and obtain the valgrind output by running the command below: 

```
valgrind --leak-check=full --show-leak-kinds=all ./examples/z_pub -m peer -l udp/224.0.0.13:8900#iface=lo -n 120 -v "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"
```

Note the process needs to be interrupted for the memory build-up to be observed. Allowing the process to terminate normally results in the memory being cleaned up before the program exits.

The Valgrind output from this run is given below:

```
==273731== Invalid free() / delete / delete[] / realloc()
==273731==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==273731==    by 0x109A5B: main (z_pub.c:117)
==273731==  Address 0x1ffefff40a is on thread 1's stack
==273731== 
==273731== 
==273731== HEAP SUMMARY:
==273731==     in use at exit: 535,936 bytes in 841 blocks
==273731==   total heap usage: 910 allocs, 70 frees, 571,985 bytes allocated
==273731== 
==273731== 16 bytes in 1 blocks are definitely lost in loss record 1 of 6
==273731==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==273731==    by 0x489BD6F: z_malloc (system.c:95)
==273731==    by 0x489ADC2: __get_ip_from_iface (network.c:267)
==273731==    by 0x489AF12: _z_open_udp_multicast (network.c:299)
==273731==    by 0x487A17D: _z_f_link_listen_udp_multicast (udp.c:135)
==273731==    by 0x4879A45: _z_listen_link (link.c:101)
==273731==    by 0x48946C6: _z_new_transport_peer (manager.c:70)
==273731==    by 0x4894856: _z_new_transport (manager.c:108)
==273731==    by 0x487CCB7: __z_open_inner (session.c:48)
==273731==    by 0x487D06F: _z_open (session.c:132)
==273731==    by 0x487560A: z_open (api.c:526)
==273731==    by 0x1097E1: main (z_pub.c:78)
==273731== 
==273731== 4,800 bytes in 120 blocks are indirectly lost in loss record 2 of 6
==273731==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==273731==    by 0x489BD6F: z_malloc (system.c:95)
==273731==    by 0x488BFF3: __z_wbuf_new_iosli (iobuf.c:248)
==273731==    by 0x488C0CE: _z_wbuf_make (iobuf.c:265)
==273731==    by 0x4896FC8: _z_multicast_send_n_msg (tx.c:119)
==273731==    by 0x489392C: _z_send_n_msg (tx.c:30)
==273731==    by 0x487BCCF: _z_write (primitives.c:169)
==273731==    by 0x4875CDA: z_publisher_put (api.c:684)
==273731==    by 0x1099D0: main (z_pub.c:105)
==273731== 
==273731== 9,600 bytes in 240 blocks are indirectly lost in loss record 3 of 6
==273731==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==273731==    by 0x489BD6F: z_malloc (system.c:95)
==273731==    by 0x488BFF3: __z_wbuf_new_iosli (iobuf.c:248)
==273731==    by 0x488C6C5: _z_wbuf_write_bytes (iobuf.c:392)
==273731==    by 0x487E001: _z_bytes_val_encode (codec.c:300)
==273731==    by 0x487E082: _z_bytes_encode (codec.c:312)
==273731==    by 0x488119F: _z_push_body_encode (message.c:289)
==273731==    by 0x4883721: _z_push_encode (network.c:63)
==273731==    by 0x4885081: _z_network_message_encode (network.c:482)
==273731==    by 0x4896FE1: _z_multicast_send_n_msg (tx.c:121)
==273731==    by 0x489392C: _z_send_n_msg (tx.c:30)
==273731==    by 0x487BCCF: _z_write (primitives.c:169)
==273731== 
==273731== 172,560 bytes in 120 blocks are indirectly lost in loss record 4 of 6
==273731==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==273731==    by 0x489BD6F: z_malloc (system.c:95)
==273731==    by 0x488B3DB: __z_iosli_init (iobuf.c:43)
==273731==    by 0x488C011: __z_wbuf_new_iosli (iobuf.c:250)
==273731==    by 0x488C0CE: _z_wbuf_make (iobuf.c:265)
[... 33 lines cut ...]
==273731==    definitely lost: 3,856 bytes in 121 blocks
==273731==    indirectly lost: 532,080 bytes in 720 blocks
==273731==      possibly lost: 0 bytes in 0 blocks
==273731==    still reachable: 0 bytes in 0 blocks
==273731==         suppressed: 0 bytes in 0 blocks
==273731== 
==273731== For lists of detected and suppressed errors, rerun with: -s
==273731== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
```

### To reproduce

1. Run the ```z_pub`` example using the following command:

```
valgrind --leak-check=full --show-leak-kinds=all ./examples/z_pub -m peer -l udp/224.0.0.13:8900#iface=lo -n 120 -v "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [... cut ...] aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"
```

2. When complete the Valgrind log will show a build-up of indirectly lost memory.

### System info

Platform: Ubuntu 22.04 64-bit
CPU: Intel i7-1260P
Zenoh version: master (https://github.com/eclipse-zenoh/zenoh-pico/commit/ac566f57903cb10a0b0455247cdce4ca4f42e156)

cguimaraes, 2023-12-05:
This might be related to #284 .

gmartin82, 2023-12-06:
I've submitted [a PR](https://github.com/eclipse-zenoh/zenoh-pico/pull/289) which seems to resolve the memory leak. When applied, memory usage no longer increases over time and Valgrind no longer reports indirectly lost memory.

While investigating I also found this affects the unicast transport when fragmentation occurs.

PR #289 (2023-12-06): Clear buffer when done processing fragmented data to free the memory
This PR attempts to fix #288 by clearing the contents of the _z_wbuf_t created when processing fragmented data. With this fix memory usage no longer increases over time.
closed 2023-12-06 by Mallets