eclipse-zenoh/zenoh-pico #291: [Bug] Failure to receive large samples

gmartin82, 2023-12-06:
### Describe the bug

In client mode, I have encountered an issue with receiving samples >= 32 KB. Samples appear to be published successfully to a Zenoh Router but are not received by a subscribing application. The exact size which triggers the issue seems to vary from application to application.

### To reproduce

Start a Zenoh router:

```./target/release/zenohd -l tcp/127.0.0.1:7447 --no-multicast-scouting```

Start a Zenoh Pico subscriber:

```./build/examples/z_sub -e tcp/127.0.0.1:7447```

Start a Zenoh Pico publisher with large, randomly generated character data:

```
rand=$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 65536 | head -n 1)
./build/examples/z_pub -e tcp/127.0.0.1:7447 -v $rand
```

Observe that samples are not received by ```z_sub```.

### System info

Platform: Ubuntu 22.04 64-bit
CPU: Intel i7-1260P
Zenoh Pico version: master (https://github.com/eclipse-zenoh/zenoh-pico/commit/ae3c2376709c6e90875d49197935dbc4d5ba2105)

jean-roland, 2023-12-18:
I confirm, this looks like a regression from Zenoh starting from hop-to-hop compression.

Mallets, 2023-12-18:
The size of the RX buffer should depend on wether the link is streamed or not. In case it is streamed, we need to account also for the 2 bytes indicating the batch length. The compression PR fixed that behaviour in Zenoh Rust. However, by doing so, it broke fragmentation compatibility with Zenoh-Pico since those 2 bytes are not accounted.

As a temporary fix, I've created this PR: https://github.com/eclipse-zenoh/zenoh/pull/630

Mallets, 2023-12-18:
I've tested https://github.com/eclipse-zenoh/zenoh-pico/pull/297/commits/479c1737e1a8ed599c6def2d6c99f978af4fd64b s and it works against https://github.com/eclipse-zenoh/zenoh/commit/780ec606ba4000a17b2da0e2b0b668e11d37c65e.
Once https://github.com/eclipse-zenoh/zenoh-pico/pull/297 lands, this issue should be solved. In the meanwhile, I'm waiting to merge https://github.com/eclipse-zenoh/zenoh/pull/630.

Mallets, 2023-12-18:
#297 has been merged. Before closing this issue, @gmartin82 can you please verify on your side that everything is now working as expected?

gmartin82, 2023-12-18:
I've tried a range of sample sizes from 8 bytes to 512 Mbytes in client mode and everything appears to be working as expected with the master versions of Zenoh and Zenoh Pico.

However, I am still experiencing communication issues between publishers and subscribers in peer mode. After sending some samples successfully communication freezes up. It doesn't happen every time but seems to be happening often.

Mallets, 2023-12-18:
Peer mode on UDP multicast doesn't have any reliability. So, it's normal in case of fragmentation to loose data. Do you see data going on the network e.g. with Wireshark?

gmartin82, 2023-12-18:
Yes I can see data flowing on the network with Wireshark.

if I disable reliability Cyclone DDS is exhibiting the same behavior so I think you’re right that this is due to UDP multicast lacking reliability.

gmartin82, 2023-12-19:
I reran my tests overnight on my faster desktop machine and the results aren't looking as good as when run on my laptop. A lot of tests failed to complete. These are the results of throughput testing for Zenoh Pico in client mode:

| Sample Size | Samples/s | Mbit/s |
| ----------- | --------- | ------ |
| 8 B         | 96435.36  | 21.59  |
| 16 B        | 95239.17  | 27.42  |
| 32 B        | 93669.61  | 38.97  |
| 64 B        |           |        |
| 128 B       | 90265.66  | 106.86 |
| 256 B       |           |        |
| 512 B       |           |        |
| 1 KB        |           |        |
| 2 KB        |           |        |
| 4 KB        |           |        |
| 8 KB        |           |        |
| 16 KB       |           |        |
| 32 KB       |           |        |
| 64 KB       | 1538.54   | 806.88 |
| 128 KB      | 769.59    | 807.08 |
| 256 KB      | 370.9     | 777.88 |
| 512 KB      |           |        |
| 1 MB        |           |        |
| 2 MB        |           |        |
| 4 MB        |           |        |
| 8 MB        |           |        |
| 16 MB       |           |        |
| 32 MB       |           |        |
| 64 MB       |           |        |
| 128 MB      |           |        |
| 256 MB      |           |        |
| 512 MB      |           |        |

Zenoh commit #788172b0360caa38db24ec2d238734a2ebf40d32
Zenoh Pico commit #63168384689bc5846cf9fdb90743824768963fdd

Mallets, 2023-12-19:
I think something weird in the tests for payload < `64KB`.
For tests `> 256KB` be aware that by default Zenoh-Pico does not accept any message larger thank `300KB`. You should modify this parameter to accept larger messages: https://github.com/eclipse-zenoh/zenoh-pico/blob/a0b421ccb5fc015b6a9fa96f1f8338f389bc2c5d/include/zenoh-pico/config.h#L309

gmartin82, 2023-12-19:
I think you're right for sample sizes > 256 KB. I hadn't overridden the maximum fragment size.

jean-roland, 2023-12-22:
Should we close this issue?

gmartin82, 2024-01-09:
Closing as it worked as expected once the maximum fragment size was increased.