eclipse-zenoh/zenoh-pico #312: RefCount for zenoh entities

imstevenpmwork, 2024-01-05:
### Describe the feature

As an example, currently if we call ```z_publisher_put(...)``` after having closed the session we get a segmentation fault:
```
zp_stop_read_task(...);
zp_stop_lease_task(...);
z_close(...);
...
z_publisher_put(...)
```
This happens because ```z_publisher_t``` maintains a reference to the session, but if this one was closed, the reference is unchanged. This is only one of the examples and even thought the order of these calls doesn't make sense, our API should be robust enough to avoid these types of bad-usage.
This can be solved by implementing a reference counting mechanism for the declared zenoh entities. The architecture is still to be defined, but I'm opening this issue to keep track of it.

cguimaraes, 2024-01-06:
I believe that @p-avital implemented (or is implementing) the baseline for a refcount mechanism.

In this case, the `close` call would be postponed until no references to the session exist.

jean-roland, 2024-01-08:
As of now, refcounting is present but only used for subscriptions and queryables

imstevenpmwork, 2024-01-19:
Solved by https://github.com/eclipse-zenoh/zenoh-pico/pull/322

closed 2024-01-19 by imstevenpmwork