eclipse-zenoh/zenoh-pico #344: Review of the protocol

Mallets, 2024-02-08:
### Describe the release item

### Describe the feature

As part of the effort for the next [Zenoh 1.0.0 release](https://github.com/eclipse-zenoh/roadmap/issues/116), a full review of the existing wire protocol is desirable. 
This PR is a sister PR of https://github.com/eclipse-zenoh/zenoh/issues/673.

This includes aspects like (but not limited to):
- [ ] Entity ID 
- [ ] Encoding field
- [ ] Declarations
- [ ] Verbatim chunks impact on Declarations
- [ ] SampleKind in replies (linked to https://github.com/eclipse-zenoh/zenoh/pull/717)

Changes will be first pushed to https://github.com/eclipse-zenoh/zenoh-pico/tree/protocol_changes branch.

Mallets, 2024-07-16:
@jean-roland I believe this issue can be closed now based on the recent development in `dev/1.0.0`.

jean-roland, 2024-07-16:
Agreed, you can close it.

closed 2024-07-16 by Mallets