eclipse-zenoh/zenoh-pico #355: Struggling ESP32 serial to linux

Mallets, 2024-02-26:
### Discussed in https://github.com/eclipse-zenoh/roadmap/discussions/120

<div type='discussions-op-text'>

<sup>Originally posted by **vortex314** February 25, 2024</sup>
I got Zenohd running on linux and via USB serial FTDI chip connected to UART2 of an ESP32 board
The ESP32 board is running zenoh-pico via UART2. I use UART0 for debugging. 
It looks that the daemon and the micro-controller, see each other but the session is not opened.
At µC side 
```
 >>> Starting Zenoh-Pico Publisher...
 Opening Zenoh Session...  
␛[0;32mI (6027) uart: queue free spaces: 100␛[0m
[1970-01-01T00:00:06Z INFO ::_z_unicast_open_client] Sending Z_INIT(Syn)
[1970-01-01T00:00:06Z DEBUG ::_z_init_encode] Encoding _Z_MID_T_INIT
[1970-01-01T00:00:07Z DEBUG ::_z_keep_alive_decode] Decoding _Z_MID_T_KEEP_ALIVE
Unable to open session!
 >>> Starting Zenoh-Pico Publisher...
```
At Linux side 
```
[2024-02-25T00:19:26Z TRACE z_serial] Read 27 bytes COBS [02, 13, 18, 01, 08, F2, 4F, F5, 5A, 85, 7B, 02, F8, BC, 4A, B8, 05, 0C, 1F, 93, F1, 6B, 15, 63, 48, FB, 00]
[2024-02-25T00:19:26Z TRACE z_serial] Frame after COBS encoding [13, 00, 01, 08, F2, 4F, F5, 5A, 85, 7B, 02, F8, BC, 4A, B8, 05, 0C, 1F, 93, F1, 6B, 15, 63, 48, FB]
[2024-02-25T00:19:26Z TRACE z_serial] Received CRC FB486315  Computed CRC FB486315
[2024-02-25T00:19:26Z TRACE zenoh_transport::unicast::universal::rx] Received: TransportMessage { body: InitSyn(InitSyn { version: 8, whatami: Client, zid: 6bf1931f0c05b84abcf8027b855af54f, resolution: Resolution(10), batch_size: 65535, ext_qos: None, ext_shm: None, ext_auth: None, ext_mlink: None, ext_lowlatency: None, ext_compression: None }) }
[2024-02-25T00:19:26Z DEBUG zenoh_transport::unicast::universal::rx] Transport: 80dbb154a44a7ac05f939447a2c1dc93. Message handling not implemented: TransportMessage { body: InitSyn(InitSyn { version: 8, whatami: Client, zid: 6bf1931f0c05b84abcf8027b855af54f, resolution: Resolution(10), batch_size: 65535, ext_qos: None, ext_shm: None, ext_auth: None, ext_mlink: None, ext_lowlatency: None, ext_compression: None }) }
[2024-02-25T00:19:26Z TRACE async_std::task::builder] block_on
[2024-02-25T00:19:26Z TRACE async_io::driver] block_on()
[2024-02-25T00:19:26Z TRACE async_io::driver] block_on: completed
[2024-02-25T00:19:26Z TRACE polling] Poller::notify()
[2024-02-25T00:19:26Z TRACE polling::epoll] notify: epoll_fd=6, event_fd=7
[2024-02-25T00:19:26Z TRACE polling::epoll] new events: epoll_fd=6, res=1
[2024-02-25T00:19:26Z TRACE async_io::driver] block_on: sleep until notification
[2024-02-25T00:19:26Z TRACE polling::epoll] modify: epoll_fd=6, fd=7, ev=Event { key: 18446744073709551615, readable: true, writable: false }
[2024-02-25T00:19:26Z TRACE async_io::reactor] process_timers: 0 ready wakers
[2024-02-25T00:19:26Z TRACE async_io::reactor] react: 0 ready wakers
[2024-02-25T00:19:26Z TRACE async_io::driver] main_loop: sleeping for 50 us
[2024-02-25T00:19:26Z TRACE async_io::driver] main_loop: notified
[2024-02-25T00:19:26Z TRACE async_io::driver] main_loop: waiting on I/O
[2024-02-25T00:19:26Z TRACE async_io::reactor] process_timers: 0 ready wakers
[2024-02-25T00:19:26Z TRACE polling] Poller::wait(_, Some(1.487404384s))
[2024-02-25T00:19:26Z TRACE polling::epoll] wait: epoll_fd=6, timeout=Some(1.487404384s)
[2024-02-25T00:19:26Z TRACE polling::epoll] modify: epoll_fd=6, fd=8, ev=Event { key: 18446744073709551615, readable: true, writable: false }
```
Source µC : https://github.com/vortex314/zenoh-projects/blob/main/zenoh-espidf/src/main.cpp
zenohd config : https://github.com/vortex314/zenoh-projects/blob/main/zenohd/zenoh.json5
Build flags for zenohd : https://github.com/vortex314/zenoh-projects/blob/main/zenohd/zenoh.json5
</div>

Mallets, 2024-02-26:
Tagging @jean-roland

vortex314, 2024-03-06:
There was no timeout in the read at pico side. has been corrected in the pull request.

cguimaraes, 2024-03-06:
Related to #140

jean-roland, 2024-04-04:
Solved in #383

closed 2024-04-04 by Mallets