eclipse-zenoh/zenoh-pico #357: [Bug]  Serial transport gets closed after 1 connection on router, client side  also doesn´t do any retry

vortex314, 2024-02-26:
### Describe the bug

When connecting via serial port to zenohd, it accepts the session , transports data and the closes the connection.

From that point on it ignores any new connection attempts.

```
]
[2024-02-26T21:00:09Z TRACE z_serial] Frame after COBS encoding [2B, 00, 05, 63, DD, 01, 21, 0D, 01, 23, 5B, 20, 20, 39, 38, 5D, 20, 5B, 45, 53, 50, 49, 44, 46, 5D, 7B, 45, 53, 50, 33, 32, 7D, 20, 20, 5A, 65, 6E, 6F, 68, 2D, 50, 69, 63, 6F, 21, AC, F2, 19, FF]
[2024-02-26T21:00:09Z TRACE z_serial] Received CRC FF19F2AC  Computed CRC FF19F2AC
[2024-02-26T21:00:09Z TRACE zenoh_transport::unicast::universal::rx] Received: TransportMessage { body: Frame(Frame { reliability: BestEffort, sn: 99, ext_qos: QoSType { inner: 5 }, payload: [NetworkMessage { body: Push(Push { wire_expr: 1:Sender:, ext_qos: QoS { priority: Data, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, payload: Put(Put { timestamp: None, encoding: Exact(Empty), ext_sinfo: None, ext_attachment: None, ext_unknown: [], payload: ZBuf { slices: [[5b, 20, 20, 39, 38, 5d, 20, 5b, 45, 53, 50, 49, 44, 46, 5d, 7b, 45, 53, 50, 33, 32, 7d, 20, 20, 5a, 65, 6e, 6f, 68, 2d, 50, 69, 63, 6f, 21]] } }) }) }] }) }
[2024-02-26T21:00:10Z TRACE z_serial] Read 51 bytes COBS [02, 2B, 30, 05, 64, DD, 01, 21, 0D, 01, 23, 5B, 20, 20, 39, 39, 5D, 20, 5B, 45, 53, 50, 49, 44, 46, 5D, 7B, 45, 53, 50, 33, 32, 7D, 20, 20, 5A, 65, 6E, 6F, 68, 2D, 50, 69, 63, 6F, 21, FA, B4, F0, F9, 00]
[2024-02-26T21:00:10Z TRACE z_serial] Frame after COBS encoding [2B, 00, 05, 64, DD, 01, 21, 0D, 01, 23, 5B, 20, 20, 39, 39, 5D, 20, 5B, 45, 53, 50, 49, 44, 46, 5D, 7B, 45, 53, 50, 33, 32, 7D, 20, 20, 5A, 65, 6E, 6F, 68, 2D, 50, 69, 63, 6F, 21, FA, B4, F0, F9]
[2024-02-26T21:00:10Z TRACE z_serial] Received CRC F9F0B4FA  Computed CRC F9F0B4FA
[2024-02-26T21:00:10Z TRACE zenoh_transport::unicast::universal::rx] Received: TransportMessage { body: Frame(Frame { reliability: BestEffort, sn: 100, ext_qos: QoSType { inner: 5 }, payload: [NetworkMessage { body: Push(Push { wire_expr: 1:Sender:, ext_qos: QoS { priority: Data, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, payload: Put(Put { timestamp: None, encoding: Exact(Empty), ext_sinfo: None, ext_attachment: None, ext_unknown: [], payload: ZBuf { slices: [[5b, 20, 20, 39, 39, 5d, 20, 5b, 45, 53, 50, 49, 44, 46, 5d, 7b, 45, 53, 50, 33, 32, 7d, 20, 20, 5a, 65, 6e, 6f, 68, 2d, 50, 69, 63, 6f, 21]] } }) }) }] }) }
[2024-02-26T21:00:10Z TRACE z_serial] Read 13 bytes COBS [02, 05, 0A, 05, 65, 1E, 01, 01, 08, FA, 37, FA, 00]
[2024-02-26T21:00:10Z TRACE z_serial] Frame after COBS encoding [05, 00, 05, 65, 1E, 01, 01, 08, FA, 37, FA]
[2024-02-26T21:00:10Z TRACE z_serial] Received CRC FA37FA08  Computed CRC FA37FA08
[2024-02-26T21:00:10Z TRACE zenoh_transport::unicast::universal::rx] Received: TransportMessage { body: Frame(Frame { reliability: BestEffort, sn: 101, ext_qos: QoSType { inner: 5 }, payload: [NetworkMessage { body: Declare(Declare { ext_qos: QoS { priority: Data, congestion: Drop, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, body: UndeclareKeyExpr(UndeclareKeyExpr { id: 1 }) }) }] }) }
[2024-02-26T21:00:10Z TRACE z_serial] Read 10 bytes COBS [02, 02, 02, 23, FD, 53, 76, 8F, F9, 00]
[2024-02-26T21:00:10Z TRACE z_serial] Frame before COBS encoding [01, 00, 04, BA, 6C, 6A, F8]
[2024-02-26T21:00:10Z TRACE z_serial] Wrote 9bytes COBS [02, 01, 06, 04, BA, 6C, 6A, F8, 00]
[2024-02-26T21:00:13Z TRACE z_serial] Frame before COBS encoding [01, 00, 04, BA, 6C, 6A, F8]
[2024-02-26T21:00:13Z TRACE z_serial] Wrote 9bytes COBS [02, 01, 06, 04, BA, 6C, 6A, F8, 00]
[2024-02-26T21:00:15Z TRACE z_serial] Frame before COBS encoding [01, 00, 04, BA, 6C, 6A, F8]
[2024-02-26T21:00:15Z TRACE z_serial] Wrote 9bytes COBS [02, 01, 06, 04, BA, 6C, 6A, F8, 00]
[2024-02-26T21:00:18Z TRACE z_serial] Frame before COBS encoding [01, 00, 04, BA, 6C, 6A, F8]
[2024-02-26T21:00:18Z TRACE z_serial] Wrote 9bytes COBS [02, 01, 06, 04, BA, 6C, 6A, F8, 00]
[2024-02-26T21:00:20Z DEBUG zenoh_transport::unicast::universal::link] serial//dev/ttyUSB2 => serial/958225c9-cda3-46c8-b947-5d2aa165a666:BatchConfig { mtu: 1500, is_streamed: false }: expired after 10000 milliseconds at io/zenoh-transport/src/unicast/universal/link.rs:290.
[2024-02-26T21:00:20Z DEBUG zenoh_transport::unicast::universal::transport] [8aa074711b0ec1ae659c86d14886d757] Closing transport with peer: c4aa8876ec0c09676d6071a6def58a03
[2024-02-26T21:00:20Z TRACE zenoh_transport::unicast::universal::link] serial//dev/ttyUSB2 => serial/958225c9-cda3-46c8-b947-5d2aa165a666: closing
[2024-02-26T21:00:25Z TRACE zenoh_transport::unicast::manager] Accepting link... serial//dev/ttyUSB2 => serial/958225c9-cda3-46c8-b947-5d2aa165a666
[2024-02-26T21:00:25Z TRACE z_serial] Read 5 bytes COBS [28, F9, E1, FB, 00]
[2024-02-26T21:00:35Z DEBUG zenoh_transport::unicast::manager] future has timed out

```

Also from client side on the serial port , there is no retry attempt if the setup of the session fails.
It keep hanging on a read and never times out. 

```
I (312) cpu_start: Chip rev:         v1.0
I (317) heap_init: Initializing. RAM available for dynamic allocation:
I (324) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (330) heap_init: At 3FFB3190 len 0002CE70 (179 KiB): DRAM
I (336) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (342) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (349) heap_init: At 4008CFE8 len 00013018 (76 KiB): IRAM
I (356) spi_flash: detected chip: winbond
I (360) spi_flash: flash io: dio
I (365) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
 >>> Starting Zenoh-Pico Publisher...
 Opening Zenoh Session...  
I (485) uart: queue free spaces: 100
[1970-01-01T00:00:00Z INFO ::_z_unicast_open_client] Sending Z_INIT(Syn)
[1970-01-01T00:00:00Z DEBUG ::_z_init_encode] Encoding _Z_MID_T_INIT
```
Restarting the client doesn´t help also as the zenohd completely ignores the port.

What am I missing ? The way server and client handle a serial unreliable connection simply cannot work. 

### To reproduce

- start zenohd zith serial listen port
- start client serial on esp32
- first session successful
- second session fails at server and client side 

### System info

- code : https://github.com/vortex314/zenoh-projects

vortex314, 2024-03-08:
While testing with zenohd , I got this panic call :
```
Clieven@pcpav2:~/workspace/zenoh$ ./zenoh.sh
rm: cannot remove 'strace3.log': No such file or directory
[2024-03-08T17:38:56Z INFO  zenohd] zenohd v0.11.0-dev-101-gf62ff682-modified built with rustc 1.72.0 (5680fa18f 2023-08-23)
[2024-03-08T17:38:56Z INFO  zenohd] Initial conf: {"adminspace":{"permissions":{"read":true,"write":false}},"aggregation":{"publishers":[],"subscribers":[]},"connect":{"endpoints":[]},"id":"d8094922038a630b176fec1f18275c8","listen":{"endpoints":["serial//dev/ttyUSB1#baudrate=115200","tcp/localhost:7447"]},"metadata":{"location":"Penny Lane","name":"strawberry"},"mode":"router","plugins":{},"plugins_search_dirs":[],"queries_default_timeout":10000,"routing":{"peer":{"mode":"peer_to_peer"},"router":{"peers_failover_brokering":true}},"scouting":{"delay":200,"gossip":{"autoconnect":{"peer":"router|peer","router":""},"enabled":true,"multihop":false},"multicast":{"address":"224.0.0.224:7446","autoconnect":{"peer":"router|peer","router":""},"enabled":true,"interface":"auto","listen":true},"timeout":3000},"timestamping":{"drop_future_timestamp":false,"enabled":{"client":false,"peer":false,"router":true}},"transport":{"auth":{"pubkey":{"key_size":null,"known_keys_file":null,"private_key_file":null,"private_key_pem":null,"public_key_file":null,"public_key_pem":null},"usrpwd":{"dictionary_file":null,"password":null,"user":null}},"link":{"protocols":null,"rx":{"buffer_size":65535,"max_message_size":1073741824},"tls":{"client_auth":false,"client_certificate":null,"client_private_key":null,"root_ca_certificate":null,"server_certificate":null,"server_name_verification":null,"server_private_key":null},"tx":{"batch_size":65535,"keep_alive":4,"lease":10000,"queue":{"backoff":100,"size":{"background":4,"control":1,"data":4,"data_high":2,"data_low":4,"interactive_high":1,"interactive_low":1,"real_time":1}},"sequence_number_resolution":"32bit","threads":2},"unixpipe":{"file_access_mask":null}},"multicast":{"compression":{"enabled":false},"join_interval":2500,"max_sessions":1000,"qos":{"enabled":false}},"shared_memory":{"enabled":true},"unicast":{"accept_pending":100,"accept_timeout":10000,"compression":{"enabled":false},"lowlatency":false,"max_links":10,"max_sessions":1000,"qos":{"enabled":true}}}}
[2024-03-08T17:38:56Z INFO  zenoh::net::runtime] Using PID: d8094922038a630b176fec1f18275c8
[2024-03-08T17:38:56Z INFO  zenoh::net::runtime::orchestrator] Zenoh can be reached at: tcp/127.0.0.1:7447
[2024-03-08T17:38:56Z INFO  zenoh::net::runtime::orchestrator] Zenoh can be reached at: serial//dev/ttyUSB1
[2024-03-08T17:38:56Z INFO  zenoh::net::runtime::orchestrator] zenohd listening scout messages on 224.0.0.224:7446
[2024-03-08T17:38:56Z INFO  zenohd] Finished loading plugins
thread 'async-std/runtime' panicked at 'explicit panic', io/zenoh-transport/src/unicast/manager.rs:751:17
stack backtrace:
   0: rust_begin_unwind
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panicking.rs:593:5
   1: core::panicking::panic_fmt
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/panicking.rs:67:14
   2: core::panicking::panic
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/panicking.rs:117:5
   3: async_task::raw::RawTask<F,T,S,M>::run
   4: async_global_executor::threading::thread_main_loop
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
./zenoh.sh: line 10: 2347944 Aborted                 (core dumped) ./target/release/zenohd --config=./zenoh.json5
lieven@pcpav2:~/workspace/zenoh$ 
```

jean-roland, 2024-04-04:
Related to #355
Solved in #383

closed 2024-04-04 by Mallets