eclipse-zenoh/zenoh-pico #369: [Bug] [ESP-IDF] Unicast lease (possibly others?) does not kill session on timeout

meganukebmp, 2024-03-09:
### Describe the bug

Whenever the keepalive lease times out (for example, zenohd shut down), the session never actually gets closed. Doing `z_check(session)` keeps returning true as well. The session remains active so for example a publisher will keep publishing to nothing.

### To reproduce

1. Open a session to zenohd
2. Check if the session is up
3. Close session
4. z_check(session) returns true after timeout.

### System info

Zenohd 0.10.1
zenoh-pico 0.10.1
esp32-eth01, esp32-lolin

meganukebmp, 2024-03-09:
Dissecting this issue further. Log states session will close:
https://github.com/eclipse-zenoh/zenoh-pico/blob/main/src/transport/unicast/lease.c#L56

Then disables the lease loop. It then calls this:
https://github.com/eclipse-zenoh/zenoh-pico/blob/main/src/transport/unicast/transport.c#L281

That then simply creates a close message and transmits it to the router (if it exists, generic session close command)
https://github.com/eclipse-zenoh/zenoh-pico/blob/main/src/transport/unicast/transport.c#L272
But at no point does it actually close the session on its own.

meganukebmp, 2024-03-19:
Confirmed, this issue is present on all platforms.

Easy test on linux machine:
1. Run `zenohd -l tcp/192.168.1.100:7447 -l udp/192.168.1.100:7447`
2. Build the examples of zenoh-pico
3. Run `z_pub -n 999 -m client -e tcp/192.168.1.100:7447
4. Wait for session to connect and begin publishing messages
5. Kill the `zenohd` router
6. `z_pub` keeps publishing data to nothing forever

jean-roland, 2024-03-20:
Related to #333