eclipse-zenoh/zenoh-pico #472: Add missed z_view_keyexpr_* methods

sashacmc, 2024-06-27:
### Describe the release item

 - z_view_keyexpr_from_string_autocanonize [include/zenoh_commons.h#L3541](https://github.com/eclipse-zenoh/zenoh-c/blob/6ba997f5a91c55ddc7f849c852d50bb220fcb408/include/zenoh_commons.h#L3541)
 - z_view_keyexpr_from_substring [include/zenoh_commons.h#L3567](https://github.com/eclipse-zenoh/zenoh-c/blob/6ba997f5a91c55ddc7f849c852d50bb220fcb408/include/zenoh_commons.h#L3567)
 - z_view_keyexpr_from_substring_autocanonize [include/zenoh_commons.h#L3581](https://github.com/eclipse-zenoh/zenoh-c/blob/6ba997f5a91c55ddc7f849c852d50bb220fcb408/include/zenoh_commons.h#L3581)
 - z_view_keyexpr_from_substring_unchecked [include/zenoh_commons.h#L3596](https://github.com/eclipse-zenoh/zenoh-c/blob/6ba997f5a91c55ddc7f849c852d50bb220fcb408/include/zenoh_commons.h#L3596)

sashacmc, 2024-07-02:
z_view_keyexpr_from_str_autocanonize - done

z_view_keyexpr_from_substr* - can't be done without significant rework, currently keyexpr _suffix stored as pointer to null terminated string, to allow creating view_keyexpr from substr it should be changed to store _z_string_t.

jean-roland, 2024-08-13:
@milyin could you please assign it to me?

PR #605 (2024-08-27): Add missing view_keyexpr methods
Closes #472 and #522

Rename some internal z_string from memory ownership clarity.
closed 2024-08-27 by milyin