eclipse-zenoh/zenoh-pico #476: Replace query session pointer to refcounted session.

jean-roland, 2024-06-28:
### Describe the feature

As we removed refcounting on queries, we no longer have the issue of cross-referencing rc that prevented us from refcounting the query session pointer.

A session rc is preferable to a pointer, especially if the query is used out of the callback context and more in line with the rest of the codebase.

DenisBiryukov91, 2024-06-28:
Is this really a good idea ? Do we really need to hold on session while query is still alive ? I.e. are we supposed to ignore z_close() and wait till all queries are resolved ?

jean-roland, 2024-06-28:
I mean the session may outlive the `z_close` only if the user wants to use the query out of the callback context to make a reply.

The reverse is just a UB/segfault if the query is used to send a reply after `z_close`. It feels like the former is better than the latter but it's open to discussion.

PR #510 (2024-07-10): Add weak references
Trying to fix #476, I realised the session reference itself was dropped by the refcount which means the query has no way of knowing if the session is still valid.

A possible solution is to introduce weak references. The query having a session weak references, it would be able to check the session validity before the whole rc container is dropped.

Bonus: Working on this, I realized the `z_clone` was not working properly and there was a consolidation mode edge case when receiving replies.
closed 2024-07-10 by jean-roland