eclipse-zenoh/zenoh-pico #477: Mutexes accessad although not initialized

philhoeb, 2024-06-28:
### Describe the bug

If _z_open() fails it can happen that _z_session_rc_drop() is called and mutexes are accessed although z_mutex_init() has never been called.
At least with FreeRTOS this leads to problems. Either asserts fail (if configured) or you end up in a hard fault when the null pointer is dereferenced.

### To reproduce

1. Do not start a Zenoh router
2. Call z_open() on STM32

### System info

- Platform: STM32 with FreeRTOS
- Zenoh: 1.0.0.3

jean-roland, 2024-07-01:
Will look into this, thanks for noticing us.

jean-roland, 2024-07-03:
Hello @philhoeb, could you just tell me what zenoh platform is used in your setup? The list can be found in `include/zenoh-pico/system/platform-common.h`.

PR #527 (2024-07-11): Add malloc value guards for freertos
Similar changes that were made to the flipper platform.
Should fix #477
closed 2024-07-11 by gabrik