eclipse-zenoh/zenoh-pico #538: Multiple definitions of the api/handlers in 1.0.0.4

AlexisTM, 2024-07-15:
### Describe the release item

I am currently looking into upgrading the Zenoh interface for PX4 to Zenoh-pico 1.0.0.4, upgrading it from Zenoh-pico 0.7. 

### Problem

There are multiple translation units in that project which includes `zenoh-pico.h`.
From `1.0` onwards, `zenoh-pico/api/handlers.h` implements the functions in the header file using macros.
This causes a link error as multiple definitions of the same functions exists. 

**Error:**

```bash
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_check':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: multiple definition of `z_ring_handler_sample_check'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_loan':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: multiple definition of `z_ring_handler_sample_loan'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_loan_mut':
zenoh_config.cpp:(.text.z_ring_handler_sample_loan_mut+0x0): multiple definition of `z_ring_handler_sample_loan_mut'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):zenoh.cpp:(.text.z_ring_handler_sample_loan_mut+0x0): first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_null':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: multiple definition of `z_ring_handler_sample_null'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_move':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: multiple definition of `z_ring_handler_sample_move'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_ring_handler_sample_drop':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: multiple definition of `z_ring_handler_sample_drop'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:157: first defined here
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: src/modules/zenoh/libmodules__zenoh.a(zenoh_config.cpp.obj): in function `z_fifo_handler_sample_check':
/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:162: multiple definition of `z_fifo_handler_sample_check'; src/modules/zenoh/libmodules__zenoh.a(zenoh.cpp.obj):/home/alexis/work/PX4-Zenoh/src/modules/zenoh/zenoh-pico/include/zenoh-pico/api/handlers.h:162: first defined here
[This goes on, it triggers for all functions of handlers.h]
```

### Ideal solution

There shouldn't be any implementation in the header file to avoid breaking the **ODR (One Definition Rule)**.

### Temporary fix

It is possible to discard the error but it is not a solution.

```cmake
target_link_libraries(modules__zenoh  PRIVATE -Wl,--allow-multiple-definition)
```

DenisBiryukov91, 2024-07-15:
It is bit strange, since those functions are marked as static inline, so there should be a separate instance of the same function for each translation unit. Any idea why error is still emitted ?

AlexisTM, 2024-07-15:
The PX4 Firmware is built in `MinSizeRel`, so my guess is the compiler just decides `inline` would take more space and discards it.

> The inline keyword tells the compiler that inline expansion is preferred. However, the compiler can ignore it.

DenisBiryukov91, 2024-07-15:
yes, but still static makes function have internal linkage, thus not visible from anywhere, except corresponding translation unit. My guess is there is some kind of conflict between extern "C" and static inline (since you are apparently using cpp). Could you confirm that you are using the  recent enough version i.e. the one which has extern "C" line in handlers.h ?

AlexisTM, 2024-07-15:
This setup does have both C and CPP. The module is built in C, the usage is in CPP.

I confirm the setup I used was the last 1.0.0.4 _with_ the `extern "C"`. Removing it still has the same issue. (Identical linking error).

Current zenoh-pico commit:
```
commit 42d6a6c0e396a716e0b2224e64b3996f06058c43 (HEAD, tag: 1.0.0.4, origin/release/1.0.0.4)
Author: eclipse-zenoh-bot <eclipse-zenoh-bot@users.noreply.github.com>
Date:   Wed Jul 10 15:52:05 2024 +0000

    chore: Bump version to 1.0.0.4
```

AlexisTM, 2024-07-15:
@DenisBiryukov91 I created a minimal reproduction branch: https://github.com/AlexisTM/zenoh-pico/tree/issue/linking_odr

Compare: https://github.com/eclipse-zenoh/zenoh-pico/compare/release/1.0.0.4...AlexisTM:zenoh-pico:issue/linking_odr

Just building the example results in:

```bash
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_check':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_check'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_loan':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_loan'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_loan_mut':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_loan_mut'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_null':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_null'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_move':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_move'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_ring_handler_sample_drop':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:157: multiple definition of `z_ring_handler_sample_drop'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:157: first defined here
/usr/bin/ld: CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub_tu2.c.o: in function `z_fifo_handler_sample_check':
/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub_tu2.c:162: multiple definition of `z_fifo_handler_sample_check'; CMakeFiles/z_sub_tu2.dir/unix/c11/z_sub.c.o:/home/alexis/work/zenoh-pico/examples/unix/c11/z_sub.c:162: first defined here
[and so on]
```

In this case, it is not an embedded target, but a plain Unix, Ubuntu 24.04 using `gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0`.

Hope it helps

gabrik, 2024-07-15:
@AlexisTM #543 should have addressed this, please check and close the issue if it solves.

AlexisTM, 2024-07-16:
That does solve the issue, thank you!

PR #543 (2024-07-15): make owned functions for handlers inline
make owned functions for handlers inline;
remove unnecessary extern C from handlers.h;
resolves #538