eclipse-zenoh/zenoh-pico #571: Fifo channel handler should behave in a blocking manner

YuanYuYuan, 2024-08-05:
### Describe the bug

SKIP

### To reproduce

SKIP

### System info

- commit: [e837b67](https://github.com/eclipse-zenoh/zenoh-pico/commit/e837b672f3412f361873ba4b780a2d01360465f6)

DenisBiryukov91, 2024-08-05:
It seems to be already the case based on https://github.com/eclipse-zenoh/zenoh-pico/blob/262b8206f508056d12700aec08614481946cd08e/src/collections/fifo_mt.c#L79. 
The only case where the messages get dropped is in case of a single thread, where blocking behavior obviously does not make any sense.

YuanYuYuan, 2024-08-06:
Hi @DenisBiryukov91, thanks for your clarification. I'm just unaware of the difference between single-threaded and multi-threaded versions. I saw `_z_fifo_push_drop` and thought we only support a dropping push. https://github.com/eclipse-zenoh/zenoh-pico/blob/262b8206f508056d12700aec08614481946cd08e/src/collections/fifo.c?plain=1#L38 

Then let's close this issue and I will update this part in the migration guide accordingly.