eclipse-zenoh/zenoh-pico #579: `_Z_OWNED_FUNCTIONS_SYSTEM_IMPL` macro is dummy

milyin, 2024-08-07:
### Describe the release item

This macro implements common functions like `z_check`, `z_null` etc. But as this macro is used on structure with `_val` field of unknown (platform-dependent) type, like `_z_task_t` which is defined differently for each platform, this macro just provides empy implementation for these functions.
This makes our API incorrect. E.g. it's guaranteed that `z_drop` after `z_null` is safe, as well as double `z_drop`. But this is not the case for this implementation.

This could be one more argument for getting rid of `z_null` / `z_check` and removing guarantees of double-drop. See discussion in https://github.com/eclipse-zenoh/zenoh-c/issues/492

sashacmc, 2024-08-08:
@milyin please assign it to me

milyin, 2024-08-08:
The proposed solution is just don't support double drop for these types. This is necessary for C++ wrapper only, but these types are not supposed to be used in Zenoh C++ - the C++ have it's own support for these primitives (thread, mutex, etc).

The only thing to do then is just to remove the dummy functions z_null / z_check for these types. This will be the signal that these types are specific and doesn't have "gravestone" state

closed 2024-08-08 by milyin