eclipse-zenoh/zenoh-pico #634: Upstream as module to Zephyr RTOS

sutyum, 2024-09-10:
### Describe the feature

Is your feature request related to a problem?
I was not able to figure out how to use Zenoh pico in a ZephyrRTOS based firmware application. Supporting Zephyr would allow it to be used in a wide variety of boards and controllers.

Describe the solution you'd like
It seems zenoh-pico already supports various features on the ZephyrRTOS platform. Adding zenoh-pico as a module to zephyr would allow it to be built as a part of a zephyr app just by setting the relevant kconfig variables.

Describe alternatives you've considered
A blog/tutorial/example around this could be an alternative. Currently the code in the zephyr directory in the examples directory of this repo does not have a complete zephyr example.

jean-roland, 2024-09-10:
Hello @sutyum. To test zephyr integration internally, we use PlatformIo with a .ini file as simple as:
```
[env:nucleo_f767zi]
platform = ststm32
board = nucleo_f767zi
framework = zephyr
lib_deps = https://github.com/eclipse-zenoh/zenoh-pico#main
```

Now some users opened a PR to more directly integrate zenoh-pico with Zephyr/kconfig (#395) but right now we don't have any resources to give you on how to such integration.

WilliamGFish, 2024-11-11:
Hi, I've been looking at zenoh in Zephyr over the last couple of weeks and it could be very straight forward with a few exceptions.

This pico code base since launching v1.0 it has some issues that are being ironed out but generally pulling the library in and compiling fairly painless.

I have found that there appears to be an inability to hardware TCP/IP (Offloaded networking) as the zenoh platform (I assume is in essence) is integrated directly into the software IP stack. I say this based on the code, which I guess is opening a session :
`_z_session_rc_t zsrc = _z_session_rc_new(s);` api.c

Running the publish example I always get 
> getaddrinfo error EAI_MEMORY

I am happy to help get things working.
Billy..