eclipse-zenoh/zenoh-pico #805: Zenoh Pico with Peer Functionality in Embedded Systems

WilliamGFish, 2024-11-21:
### Describe the feature

**Enhance Zenoh Pico with Peer Functionality for Broader Applicability in Embedded Systems**

Addition of peer functionality to Zenoh Pico to significantly enhance its utility in embedded systems, particularly in the sensor network and industrial controls markets. This extension aligns with the growing demand for efficient, reliable, and cost-effective data transport solutions in the Industrial Internet of Things (IIoT) space, driven by advancements in low-power, low-cost, high-throughput wireless networking technologies.

Competing solutions like WirePas, IO-Link Wireless, OPC-UA, and HART have sought to address similar challenges, but there remains a need for a more versatile and lightweight alternative that supports the complexity of industrial data requirements, including distribution, quantity, quality, and frequency of updates.

Current industrial practices often rely on intricate solutions to extract, store, and manage data in rigid systems, limiting flexibility and accessibility. However, the emerging trend toward a Universal Namespace (UNS) offers a more dynamic approach, enabling flexible data storage, discovery, reporting, and maintenance.

Integrating peer functionality into Zenoh Pico would position it as a key enabler for UNS principles, simplifying data discovery and communication in diverse environments without resorting to proprietary or overly complex systems. This enhancement would make Zenoh Pico a powerful tool for addressing the evolving needs of industrial and embedded applications.

Mallets, 2024-11-21:
Zenoh-Pico already supports peer mode over UDP multicast as explained in the [README](https://github.com/eclipse-zenoh/zenoh-pico?tab=readme-ov-file#34-basic-pubsub-example---p2p-over-udp-multicast).

Besides that, could you elaborate more on what particular use case would the peer functionality be an improvement?
In other words, which use case will suffer from using a router and Zenoh-Pico in client mode?
Looking forward to get some more insights on the underlying scenario.

XxChang, 2025-01-10:
Hello, are there a plan to support peer mode over Serial

sashacmc, 2025-01-10:
Hello @XxChang, this is not on the roadmap at the moment, but you may want to reach out ZettaScale at [contact@zettascale.tech](mailto:contact@zettascale.tech) since more options are available for the commercial version of Zenoh.