eclipse-zenoh/zenoh-pico #810: will zenoh support Custom Transport like microros?

clippinglee, 2024-11-22:
### Describe the feature

only ethernet for some OS now, will zenoh-pico will support more transport? if custom can defined by himself,it will be great

clippinglee, 2024-11-29:
z_result_t _z_open_link(_z_link_t *zl, _z_string_t *locator);if we can change define this function, we can get more transports.z_result_t _z_open_link(_z_link_t *zl, _z_string_t *locator,    _z_f_link_open _open_f,
    _z_f_link_listen _listen_f,
    _z_f_link_close _close_f,
    _z_f_link_write _write_f,
    _z_f_link_write_all _write_all_f,
    _z_f_link_read _read_f,
    _z_f_link_read_exact _read_exact_f,
    _z_f_link_free _free_f)。     so we can defined transport by custom, We can generate a library file and move it to any platform.

sashacmc, 2024-12-02:
Hello @clippinglee,
Thank you for your suggestion, it's an interesting idea, we'll think about it.

k-yokoyama-esol, 2024-12-04:
@clippinglee 
Native micro-ROS custom transport needs Agent and Client.
However, Zenoh (`rmw_zenoh`) and Zenoh-pico can communicate without the agent node.
You can see the same discussion at #2.

k-yokoyama-esol, 2024-12-04:
I was mistaken.
Your suggestion is an extension of the transport layer of Zenoh-pico. That sounds very interesting.

If we can extend OpenAMP in the same way as serial, heterogeneous cores can communicate.
There is a paper that combines micro-ROS and OpenAMP, but the existence of the XRCE-DDS Agent is a bottleneck.
<https://sasimi.jp/new/sasimi2024/files/archive/pdf/p26_R1-5.pdf>