eclipse-zenoh/zenoh-pico #836: [Bug] Compilation error: "error: conflicting declaration of C function"

abishekgoda, 2024-12-18:
### Describe the bug

I am trying to compile zenoh-pico to run on esp8266 (esp01 board). I am following along the readme file in the project to do this. I am using PlatformIO Core, version 6.1.16 on a m1 mac running macos 15.2. When I try `pio run` I get a bunch of conflicting declaration errors. It looks like an overloaded function but the compiler is "C" based. Or maybe am interpreting it wrong. How do I proceed?

### To reproduce

1. Use the example in the README.md
2. Use the following platformio.ini:
```[env:esp01]
platform = espressif8266
board = esp01
framework = arduino
lib_deps = arduino-libraries/WiFi@^1.2.7
build_flags = -DZENOH_ARDUINO_ESP32 ```
3. pio run

### System info

- Mac OS 15.2 / M1 Mac.
- ESP01 board with 1M memory. Am choosing basic 512K option

jean-roland, 2024-12-18:
Can you try with this platformio:
```
[env:esp32thing_plus]
platform = espressif8266
board = esp01
framework = arduino
lib_deps = 
	arduino-libraries/WiFi@^1.2.7
	zenoh-pico
build_flags = 
	-DZENOH_COMPILER_GCC
	-DZENOH_ARDUINO_ESP32 
```

abishekgoda, 2024-12-19:
No. Still the same. 
[err.log](https://github.com/user-attachments/files/18190647/err.log) has the full set of errors. Thanks!

jean-roland, 2024-12-19:
You're using a non-arduino example with an example framework.

closed 2024-12-19 by jean-roland

abishekgoda, 2024-12-19:
That is odd. I am following the README verbatim. And I don't even know what freertos is :-( . I'll check with the discord.

jean-roland, 2024-12-19:
Sadly, it's highly probable there's missing / obsolete info in the doc, we're working on improving it.