eclipse-zenoh/zenoh-pico #844: How can I use Zenoh-router on an operating system that does not support Docker/Rust?

yuyuyuyi2001, 2024-12-24:
### Describe the feature

I want to implement Zenoh Pico on sylixOS through cross compilation, but currently the compiled result only has one library. Is this fully functional and correct? 
[image]

Also, I would like to conduct some example validations, but SylixOS does not support Rust and Docker. Is there any other way to run Zenoh router? Thank you!

jean-roland, 2025-01-06:
First, I think your problem is more suited to the z-support channel of our [discord server](https://discord.com/channels/914168414178779197/940584045287460885) than a github issue.

Now only having the library is the expected outcome, you can then import the library in your projects and call our api to use zenoh (pico) in your application. 

You have two choice, you can test your pico on sylixOS with another pico on sylixOS with peer to peer or you can run a router on a rust-able OS on your network so you can test client mode.

closed 2025-01-06 by jean-roland

yuyuyuyi2001, 2025-01-12:
thanks you， I tried run zeonh-router in Linux， it was succeed.