eclipse-zenoh/zenoh-pico #846: [Bug] Lease task creates a busy loop on ESP32

vortex314, 2024-12-29:
### Describe the bug

When the freeRTOS ticks/sec is at 100 ( the default value ), the z_sleep_ms doesn´t suspend the thread and creates starvation for the other threads

```
z_result_t z_sleep_ms(size_t time) {
    z_time_t start = z_time_now();

    // Most sleep APIs promise to sleep at least whatever you asked them to.
    // This may compound, so this approach may make sleeps longer than expected.
    // This extra check tries to minimize the amount of extra time it might sleep.
    while (z_time_elapsed_ms(&start) < time) {
        vTaskDelay(1 / portTICK_PERIOD_MS);   // the argument becomes 0. 
    }

    return 0;
}
```
should be replaced with 
```
z_result_t z_sleep_ms(size_t time) {
    vTaskDelay ( time / portTICK_PERIOD_MS == 0 ? 1 : time / portTICK_PERIOD_MS );
    return 0;

}
```

### To reproduce

Use sample p_sub sources for espidf

### System info

Any ESP32 with ESPIDF

vortex314, 2024-12-30:
Before 
[image]
After
[image]

vortex314, 2025-01-01:
See also Discord : https://discord.com/channels/914168414178779197/940584045287460885/1323019008080871465

Mallets, 2025-01-06:
#848 has been merged. Closing the issue. Feel free to re-open it if the problem still persists.

closed 2025-01-06 by Mallets

vortex314, 2025-01-06:
@sashacmc  @Mallets   Thank you all for the quick follow-up. Great product and development !