eclipse-zenoh/zenoh-pico #861: [Bug] _z_unicast_handle_transport_message leaks memory

pdematteis, 2025-01-27:
### Describe the bug

The function `_z_unicast_handle_transport_message()` in transport/unicast/rx.c, while handling out-of-order _Z_MID_T_FRAME messages, leaks memory.
The following code should be added to cleanup dropped messages:
```
for (size_t i = 0; i < _z_svec_len(&t_msg->_body._frame._messages); i++) {
  _z_msg_clear(_z_network_message_svec_get(&t_msg->_body._frame._messages, i));
}
```

### To reproduce

.

### System info

 .

sashacmc, 2025-01-27:
@pdematteis, thanks, we'll look into it.

closed 2025-01-30 by milyin

pdematteis, 2025-01-31:
I'm sorry but the proposed solution still leak memory when messages are dropped:

By calling `_z_t_msg_frame_clear()` the executed code is:

```
void _z_t_msg_frame_clear(_z_t_msg_frame_t *msg) {
    if (!msg->_messages._aliased) {
        _z_network_message_svec_clear(&msg->_messages);
    }
}
```

and `_messages._aliased` is true, while the normal message processing is done by:

```
size_t len = _z_svec_len(&t_msg->_body._frame._messages);
  for (size_t i = 0; i < len; i++) {
      _z_network_message_t *zm = _z_network_message_svec_get(&t_msg->_body._frame._messages, i);
      zm->_reliability = tmsg_reliability;
      _z_handle_network_message(ztu->_common._session, zm, _Z_KEYEXPR_MAPPING_UNKNOWN_REMOTE);
  }
```

where `_z_handle_network_message()` finally always calls `_z_msg_clear(msg)`.

jean-roland, 2025-01-31:
Yes you're right, that's why `_z_t_msg_frame_clear` has been patched to handle aliased messages in #873.

pdematteis, 2025-01-31:
Got it! Sorry I missed that change.

PR #873 (2025-01-30): Fix memory leaks
Fixes #861 and an additional leak related to lru cache.