eclipse-zenoh/zenoh-pico #864: [Bug] _z_common_transport_clear() should call _z_task_free()

pdematteis, 2025-01-28:
### Describe the bug

`_z_common_transport_clear()`  in transport/common/transport.c call `z_free()` to cleanup tasks but `_z_task_free()` should be called instead, I think.
`z_free()` only releases memory, while  `_z_task_free()` also close handles and other platform-specific resources, for example it calls `vEventGroupDelete(ptr->join_event)` in FreeRTOS.

### To reproduce

.

### System info

.

sashacmc, 2025-01-28:
@pdematteis, thanks, we'll look into it.

pdematteis, 2025-01-29:
Just found that by calling `_z_task_free()` a little change is needed to make Windows implementation working properly:

```
z_result_t _z_task_detach(_z_task_t *task) {
    z_result_t ret = _Z_RES_OK;
    CloseHandle(*task);
    *task = NULL; // <== avoid closing the handle multiple times
    return ret;
}
```

Will check Linux as well.

closed 2025-02-04 by yellowhatter